HKUST-Aerial-Robotics/Fast-Planner #1: Cannot launch node of Type[so3_disturbance_generator/so3_disturbance_generator]

LIANGXINKAI, 2019-07-11:
Thanks for your work,but I'm trouble in something like this
can u help to solve this Problem

[image]

ZbyLGsc, 2019-07-11:
@LIANGXINKAI this node is not complied successfully. I would recommend you to check the compilation of so3_disturbance_generator. Firstly, check the CMakelist of the so3_disturbance_generator and uncomment the lines "add_executable(so3_disturbance_generator…)" . Then run catkin_make again to compile this node. Finally, make sure that in catkin_ws/devel/lib the so3_disturbance_generator exists. If not, it may lie in catkin_ws/build, just copy it to devel/lib. I will fix this problem soo :)

LIANGXINKAI, 2019-07-12:
thanks for your reply

michaelchi08, 2019-07-16:
Hello,

I followed the suggested step(uncommenting the add_exec(so3_...)) above but it doesn't compile.
This is my error msg. Please let me know how i can fix this. Thanks.  

so3_disturbance_generator.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_]+0x647): undefined reference to `ros::console::setLogLocationLevel(ros::console::LogLocation*, ros::console::levels::Level)'
so3_disturbance_generator.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_]+0x651): undefined reference to `ros::console::checkLogLocationEnabled(ros::console::LogLocation*)'
so3_disturbance_generator.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_]+0x6a3): undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, char const*, int, char const*, char const*, ...)'
so3_disturbance_generator.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_]+0x7b7): undefined reference to `ros::console::g_initialized'
so3_disturbance_generator.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_]+0x7c7): undefined reference to `ros::console::initialize()'
so3_disturbance_generator.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_]+0x816): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ros::console::levels::Level)'
so3_disturbance_generator.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_]+0x854): undefined reference to `ros::console::setLogLocationLevel(ros::console::LogLocation*, ros::console::levels::Level)'
so3_disturbance_generator.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_]+0x85e): undefined reference to `ros::console::checkLogLocationEnabled(ros::console::LogLocation*)'
so3_disturbance_generator.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_]+0x8b8): undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, char const*, int, char const*, char const*, ...)'
so3_disturbance_generator.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_]+0x8c3): undefined reference to `ros::console::g_initialized'
so3_disturbance_generator.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_]+0x8d3): undefined reference to `ros::console::initialize()'
so3_disturbance_generator.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_]+0x922): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ros::console::levels::Level)'
so3_disturbance_generator.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_]+0x960): undefined reference to `ros::console::setLogLocationLevel(ros::console::LogLocation*, ros::console::levels::Level)'
so3_disturbance_generator.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_]+0x96a): undefined reference to `ros::console::checkLogLocationEnabled(ros::console::LogLocation*)'
so3_disturbance_generator.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_]+0xa26): undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, char const*, int, char const*, char const*, ...)'
so3_disturbance_generator.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_]+0xa31): undefined reference to `ros::console::g_initialized'
so3_disturbance_generator.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_]+0xa41): undefined reference to `ros::console::initialize()'
so3_disturbance_generator.cpp:(.text._ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN13geometry_msgs8Vector3_ISaIvEEEEEvRKT_]+0xa90): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ros::console::levels::Level)'
[... cut ...]
Fast-Planner/uav_simulator/so3_disturbance_generator/CMakeFiles/so3_disturbance_generator.dir/build.make:94: recipe for target 'Fast-Planner/uav_simulator/so3_disturbance_generator/so3_disturbance_generator' failed
make[2]: *** [Fast-Planner/uav_simulator/so3_disturbance_generator/so3_disturbance_generator] Error 1
CMakeFiles/Makefile2:3849: recipe for target 'Fast-Planner/uav_simulator/so3_disturbance_generator/CMakeFiles/so3_disturbance_generator.dir/all' failed
make[1]: *** [Fast-Planner/uav_simulator/so3_disturbance_generator/CMakeFiles/so3_disturbance_generator.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

ZbyLGsc, 2019-07-16:
@michaelchi08  I just fixed some tiny compilation problems of the package. The simulator depends on Armadillo, so you should install it first (follow the updated readme). Then you can pull the latest package and try compiling it again. 

Besides, I notice that there are many many many errors related to ROS, so it is highly possible that ROS is not installed properly.

michaelchi08, 2019-07-16:
Your new cha

> @michaelchi08 I just fixed some tiny compilation problems of the package. The simulator depends on Armadillo, so you should install it first (follow the updated readme). Then you can pull the latest [...]

The changes you made fix the compilation issues. Thanks for your prompt response :)

closed 2019-07-17 by ZbyLGsc