HKUST-Aerial-Robotics/Fast-Planner #3: Add a description of the project

GigaFlopsis, 2019-08-11:
Hi guys, you have a very interesting project.
I want to try flying in the woods with your algorithm. However, your package descriptions are sorely lacking. Can we expect a description of the project in the near future?

ZbyLGsc, 2019-08-11:
@GigaFlopsis To use the planner on a real drone, the simulation.launch file is a good starting point. You only need to slightly modify the "dyn_planner/plan_manage/launch/simulation.launch" file. Please reset the topics of your map of the environment( <remap from="/laser_cloud_surround" to="/random_forest/all_map"/>) and your drone's odometry (<remap from="/odom_world" to="/state_ukf/odom"/>).

The main algorithms are implemented in the "dyn_planner".  The path_searching and bspline_opt correspond to our kinodynamic path searching and Bspline based optimization.  The plan_env implements a light weight volumetric map supporting Euclidean signed distance field used by our path searching and trajectory optimization. All of them are integrated in the plan_mange package.

In the future I will add some brief description of the main packages. (However, it is too time demanding to write a tutorial :) ). As for the details of this planner, I refer you to our related paper.

GigaFlopsis, 2019-08-11:
Thank you very much for your answer.  Also wanted to clarify what should be the TF dependency?

ZbyLGsc, 2019-08-12:
@GigaFlopsis I am not sure what you mean by "TF dependency". Can you explain it?

GigaFlopsis, 2019-08-12:
Sorry. I mean [this setup](http://wiki.ros.org/navigation/Tutorials/RobotSetup/TF)

ZbyLGsc, 2019-08-12:
@GigaFlopsis Since there isn't complex rigid body transformation for the drone, setting up TF is not necessary. You can run the simulation.launch directly. Using it on a real drone is almost the same as in the simulator.

GigaFlopsis, 2019-08-12:
Thank you.
I'm trying to move the project to ROS Melodic now , will you will support melodic in the future?

ZbyLGsc, 2019-08-12:
No plan yet. But I think currently it can also run on melodic (maybe with some tiny modification for compatibility)

closed 2019-08-12 by ZbyLGsc