HKUST-Aerial-Robotics/Fast-Planner #7: KinodynamicAstar::estimateHeuristic

zgrobot, 2019-11-18:
能回答一下，这个函数是从哪里来的吗？
const Vector3d dp = x2.head(3) - x1.head(3);//获取位置差
  const Vector3d v0 = x1.segment(3, 3);//获取x1向量中第3个元素开始的3个元素；也就是起点速度
  const Vector3d v1 = x2.segment(3, 3);

  double c1 = -36 * dp.dot(dp);
  double c2 = 24 * (v0 + v1).dot(dp);
  double c3 = -4 * (v0.dot(v0) + v0.dot(v1) + v1.dot(v1));
  double c4 = 0;
  double c5 = w_time_;

  std::vector<double> ts = quartic(c5, c4, c3, c2, c1);

  double v_max = max_vel_;
  double t_bar = (x1.head(3) - x2.head(3)).lpNorm<Infinity>() / v_max;
  ts.push_back(t_bar);

  double cost = 100000000;
  double t_d = t_bar;

  for (auto t : ts)
  {
    if (t < t_bar)
      continue;
    double c = -c1 / (3 * t * t * t) - c2 / (2 * t * t) - c3 / t + w_time_ * t;
    if (c < cost)
    {
      cost = c;
      t_d = t;
    }
  }

  optimal_time = t_d;

  return 1.0 * (1 + tie_breaker_) * cost;

GigaFlopsis, 2019-11-18:
what's the question?

zgrobot, 2019-11-19:
quartic(c5, c4, c3, c2, c1);这个是什么作用？返回的ts代表什么呢？

zgrobot, 2019-11-19:
搞清楚了，是求解一元四次方程的根

closed 2019-11-19 by zgrobot