HKUST-Aerial-Robotics/Fast-Planner #8: NonUniformBspline::getControlPointEqu3介绍

zgrobot, 2019-11-21:
能否介绍一下这个函数？
  Eigen::VectorXd prow(3), vrow(3), arow(3);
  prow << 1, 4, 1;
  vrow << -1, 0, 1;
  arow << 1, -2, 1;
以上数值是根据什么取的？或者有公式吗？

ZbyLGsc, 2019-11-21:
This is about constructing a matrix that maps the control points of B-spline to the positions of the joint of spline segments. You can refer to the general matrix representation of B-spline for details.

zgrobot, 2019-11-22:
> This is about constructing a matrix that maps the control points of B-spline to the positions of the joint of spline segments. You can refer to the general matrix representation of B-spline for details.

我知道是关于曲线的矩阵表述，但函数里把速度和加速度都放进去了，而且这个数值是经验选的，还是公式里来的？

ZbyLGsc, 2019-11-23:
@zgrobot Since we also constrain the velocity and acceleration of the trajectory at the start and end points, the matrix also includes blocks about them. The velocity and acceleration are also B-spline with lower degrees, so we can the matrix representation of the lower degree B-spline. All value can be derived in theory. You may refer to the paper "General matrix representations for b-splines".

closed 2019-11-23 by ZbyLGsc