HKUST-Aerial-Robotics/Fast-Planner #9: How to change target height?

GigaFlopsis, 2019-11-21:
Now your algorithm sets a fixed height (1m) independent of the target height. can I fix this?

ZbyLGsc, 2019-11-21:
@GigaFlopsis  You can change the height to any value you like. Please set the "wp_z" in the simulation.launch for this purpose.

qmoji, 2019-11-21:
may be this [line](https://github.com/HKUST-Aerial-Robotics/Fast-Planner/blob/master/dyn_planner/plan_manage/src/planning_fsm.cpp#L83)

ZbyLGsc, 2019-11-21:
@qmoji Actually the simulation has two different modes: FLIGHT_TYPE::PRESET_GOAL and FLIGHT_TYPE::MANUAL_GOAL. The line you marked is related to the MANUAL_GOAL mode, but does not influence the PRESET_GOAL mode, the height for the PRESET_GOAL mode is set in the .launch file.

The reason for setting the height to 1 in PRESET_GOAL mode is that, using the 3D plugin, it is not easy to choose the height accurately.  To avoid unreasonable height (caused by the inaccurate selection), I set the height to 1 directly. If you have some good ideas to set the height, feel free to change this line to use the **msg->poses[0].pose.position.z** .

GigaFlopsis, 2019-11-21:
You can try use this interactive marker https://www.youtube.com/watch?v=i1Cbbq-XMGQ
https://github.com/GigaFlopsis/ros_swarm_control/blob/px4_drone/swarm_control/src/marker_server.cpp

ZbyLGsc, 2019-11-22:
@GigaFlopsis Thanks for your suggestion. I do know that RVIZ has very similar 3D marker. I may integrate it into this project later.

closed 2019-11-23 by ZbyLGsc