HKUST-Aerial-Robotics/Fast-Planner #10: About flying height

qmoji, 2019-11-22:
This is absolute great work!
But Is it possible to limit or fix the flying height? because it plans to fly over a wall.
[image]

[image]

GigaFlopsis, 2019-11-22:
Yeah! Will be better if you add the description of planner parameters from .launch file.

ZbyLGsc, 2019-11-23:
@qmoji Theoretically this is caused by the limited sensing range of the sensor. The drone can not observed the regions overhead, and those regions are treated as collision free. In this case trajectories going over the walls, which is optimal depending on the currently available information, are generated.  

Currently you can resolve this problem by reducing the control input in z axis of the path searching (in kinodynamic_astar.cpp, um << ax, ay, az, reduce the possible values of az). This will make the drone takes less effort to change its height. Also, if you know the height of the ceiling, you can set the height in the .launch file directly (<param name="sdf_map/ceil_height" value="2.5" type="double"/>), so that region above it are treated as obstacles.

qmoji, 2019-11-24:
@ZbyLGsc thank you very much!
I set `ceil_height `as 1,  and i got this...
[image]
so i made some changes to the code:
```
for (double az = 0; az <= max_acc_ + 1e-3; az += max_acc_ * res)
{
  um << ax, ay, az;
  inputs.push_back(um);
}
```
it seems to work!
[image]

closed 2019-11-24 by qmoji