HKUST-Aerial-Robotics/Fast-Planner #12: Frequent aggresive wrong motion of quadrotor

SwiftGust, 2020-01-16:
Hi,
Thank you for sharing great work!

I have a issue in the simulation.
That it is often observed that quadrotor model follows trajectory undesirable way
that the vehicle flip towards wrong direction then return & starts to follow trajectory (See GIF below)

This easily can be observed especially when transitioning from stopped state(WAIT_TARGET) to executing new trajectory(EXEC_TRAJ) or rarely seen while following trajectory.

I'd like to know why and is this can easily be solved?

Thanks in advance.

P.S. Launch file & parameters are left untouched.
[image]

ZbyLGsc, 2020-01-16:
@SwiftGust The reason is that currently the planner does not plan the trajectory for heading (yaw angle), but just naively control the drone to head toward the direction of velocity. Therefore, if you select a goal behind the drone, it will aggressively turn back.

ZbyLGsc, 2020-01-16:
@SwiftGust The previous code does not have such a problem, because we have omnidirectional perception and we make the heading a fixed value. So we did not have the aggressive turn issue. But currently to simulate a camera with limited FoV, we should change the heading. I am writing a simple heading planner, and I will release it in a few days. This problem can be solved then.

ZbyLGsc, 2020-01-30:
@SwiftGust The heading planner is now released to enable smoother change of heading direction. It should solve the problem of aggressive wrong motion you found. You can test it and close this issue if no problem exists.

SwiftGust, 2020-01-31:
@ZbyLGsc Hi, Thank you for the update.
I have tested this and the vehicle smoothly follows B-spline trajectory any motion well.
Great work!

closed 2020-01-31 by SwiftGust