HKUST-Aerial-Robotics/Fast-Planner #14: SDF Map disappears when fusion with pointcloud topic

SwiftGust, 2020-02-05:
I'm trying to use pointcloud topic fusion method in the planner simulation.

However, I've found the critical issue that when using SDF map fusion using pointcloud + odometry method. Previously updated local map out of FOV doesn't maintained at all, only local map is updated and causes planner to fail often.

**Using depth image - local map out of FOV is well maintained**

[image]

**Using pointcloud - local map out of FOV disappears**

[image]

As far as I understood
Difference between image + camera pose fusion & pointcloud + odometry fusion is that

For Pointcloud fusion,
subscriber(indep_cloud_sub_) handles pointcloud topic with cloudCallback(...) which directly updates inflated map from pointclouds(md_.occupancy_buffer_inflate_ ) without occupancy updates.

while the depth image fusion methods
updateOccupancyCallback() calls

projectDepthImage() 
raycastProcess()
clearAndInflateLocalMap()

These 3 functions do reproject depth image to points, updates occupancy then inflates.

It seems these functions fills the same md_.occupancy_buffer_inflate_ similary with cloudCallback
but issue occurs.

So I have tested with pointcloud to update md_.proj_points_ instead of md_.occupancy_buffer_inflate_ updating directly by modifying sdf_map.cpp as follows but issue still persists but cannot find out why.

Can you guide me what is causing the issue?

**Modified followings**

**sdf_map.cpp**

**In updateOccupancyCallback, commnted projectDepthImage() for sure and prevent counter reset.**
```
void SDFMap::updateOccupancyCallback(const ros::TimerEvent& /*event*/) {
  if (!md_.occ_need_update_) return;

  /* update occupancy */
  ros::Time t1, t2;
  t1 = ros::Time::now();

  // projectDepthImage();
  raycastProcess();

  if (md_.need_clear_local_map_) {
    clearAndInflateLocalMap();
  }

  t2 = ros::Time::now();

  md_.fuse_time_ += (t2 - t1).toSec();
  md_.max_fuse_time_ = max(md_.max_fuse_time_, (t2 - t1).toSec());

  if (mp_.show_occ_time_)
    ROS_WARN("Fusion: cur t = %lf, avg t = %lf, max t = %lf", (t2 - t1).toSec(),
             md_.fuse_time_ / md_.update_num_, md_.max_fuse_time_);

  md_.occ_need_update_ = false;
  if (md_.need_clear_local_map_) md_.esdf_need_update_ = true;
  md_.need_clear_local_map_ = false;
}
```

**Modified cloudCallback(...) to update md_.proj_points_  same as proejctDepthImage()**
```
void SDFMap::cloudCallback(const sensor_msgs::PointCloud2ConstPtr& img) {

  pcl::PointCloud<pcl::PointXYZ> latest_cloud;
  pcl::fromROSMsg(*img, latest_cloud);

  md_.has_cloud_ = true;

  if (!md_.has_odom_) {
    // std::cout << "no odom!" << std::endl;
    return;
  }

  if (latest_cloud.points.size() == 0) return;

  if (isnan(md_.camera_pos_(0)) || isnan(md_.camera_pos_(1)) || isnan(md_.camera_pos_(2))) return;

  this->resetBuffer(md_.camera_pos_ - mp_.local_update_range_,
                    md_.camera_pos_ + mp_.local_update_range_);

  pcl::PointXYZ pt;
  Eigen::Vector3d p3d, p3d_inf;

  int inf_step = ceil(mp_.obstacles_inflation_ / mp_.resolution_);
  int inf_step_z = 1;

  md_.proj_points_cnt = 0;
  for (size_t i = 0; i < latest_cloud.points.size(); ++i) {
    pt = latest_cloud.points[i];
    p3d(0) = pt.x, p3d(1) = pt.y, p3d(2) = pt.z;
    md_.proj_points_[md_.proj_points_cnt++] = p3d;
  }
  md_.occ_need_update_ = true;
  md_.update_num_ +=1;
}
```

Modified launch file - cloud_simulation.launch
```
<launch>

  <!-- size of map, change the size in x, y, z according to your application -->
  <arg name="map_size_x" value="40.0"/>
  <arg name="map_size_y" value="40.0"/>
  <arg name="map_size_z" value=" 5.0"/>

  <!-- topic of your odometry such as VIO or LIO -->
  <arg name="odom_topic" value="/state_ukf/odom" />

  **<node pkg="tf" type="static_transform_publisher" name="world_map_linker" args="0 0 0 0 0 0 world map 10" />**
  <!-- main algorithm params -->
  <include file="$(find plan_manage)/launch/kino_algorithm.xml">

    <arg name="map_size_x_" value="$(arg map_size_x)"/>
    <arg name="map_size_y_" value="$(arg map_size_y)"/>
    <arg name="map_size_z_" value="$(arg map_size_z)"/>
    <arg name="odometry_topic" value="$(arg odom_topic)"/>

    <!-- camera pose: transform of camera frame in the world frame -->
    <!-- depth topic: depth image, 640x480 by default -->
    <!-- don't set cloud_topic if you already set these ones! -->
    **<arg name="camera_pose_topic" value="/pcl_render_node/camera_pose_no_use"/>
    <arg name="depth_topic" value="/pcl_render_node/depth_no_use"/>**

    <!-- topic of point cloud measurement, such as from LIDAR  -->
    <!-- don't set camera pose and depth, if you already set this one! -->
    **<arg name="cloud_topic" value="/pcl_render_node/rendered_pcl"/>**

    <!-- intrinsic params of the depth camera -->
    <arg name="cx" value="385.754"/>
    <arg name="cy" value="236.743"/>
    <arg name="fx" value="385.754"/>
    <arg name="fy" value="257.296"/>

    <!-- maximum velocity and acceleration the drone will reach -->
    <arg name="max_vel" value="2.1" />
    <arg name="max_acc" value="1.0" />

    <!-- 1: use 2D Nav Goal to select goal  -->
    <!-- 2: use global waypoints below  -->
    <arg name="flight_type" value="1" />
    
    <!-- global waypoints -->
    <!-- If flight_type is set to 2, the drone will travel these waypoints one by one -->
    <arg name="point_num" value="3" />

    <arg name="point0_x" value="19.0" />
    <arg name="point0_y" value="0.0" />
    <arg name="point0_z" value="1.0" />

    <!-- set more waypoints if you need -->
    <arg name="point1_x" value="0.0" />
    <arg name="point1_y" value="-19.0" />
    <arg name="point1_z" value="1.0" />

    <arg name="point2_x" value="0.0" />
    <arg name="point2_y" value="19.0" />
    <arg name="point2_z" value="1.0" />

  </include>

  <!-- use simulator. It should be disabled if a real drone is used -->
  <include file="$(find plan_manage)/launch/simulator.xml">
    <arg name="map_size_x_" value="$(arg map_size_x)"/>
    <arg name="map_size_y_" value="$(arg map_size_y)"/>
    <arg name="map_size_z_" value="$(arg map_size_z)"/>

     <arg name="odometry_topic" value="$(arg odom_topic)" />
  </include>

  <!-- trajectory server and trigger, you don't need to change them -->
  <node pkg="plan_manage" name="traj_server" type="traj_server" output="screen">
    <remap from="/position_cmd" to="planning/pos_cmd"/>

    <remap from="/odom_world" to="$(arg odom_topic)"/>
    <param name="traj_server/time_forward" value="1.5" type="double"/>
  </node>

  <node pkg="waypoint_generator" name="waypoint_generator" type="waypoint_generator" output="screen">
    <remap from="~odom" to="$(arg odom_topic)"/>        
    <remap from="~goal" to="/move_base_simple/goal"/>
    <remap from="~traj_start_trigger" to="/traj_start_trigger" />
    <param name="waypoint_type" value="manual-lonely-waypoint"/>    
  </node>

</launch>

```

ZbyLGsc, 2020-02-06:
@SwiftGust There might be some issues in cloudcallback, and I will check for it. The map fusion using depth images and camera pose is tested extensively in our experiment, and I will recommend you to use this interface if you can. However, the one using point cloud is not well tested, we just add this interface so that some PCs without GPU can still use our package.

Mayankm96, 2020-08-28:
I was taking a closer look into the source of this issue and I think it is due to the call to the `resetBuffer()` function which clears the local map:
```
this->resetBuffer(md_.camera_pos_ - mp_.local_update_range_,
                    md_.camera_pos_ + mp_.local_update_range_);
```

SwiftGust, 2020-08-29:
@Mayankm96 Yes you are correct, the exact line need to be removed then the issue goes away with the appended code.
This issue is actually solved in my end quite some time ago, so closing.

closed 2020-08-29 by SwiftGust