HKUST-Aerial-Robotics/Fast-Planner #16: A fatal error occurred while executing catkin_make:    ecl.h & geo/geo.h    There is no file or directory

Quixote-Sin, 2020-03-14:


Quixote-Sin, 2020-03-14:
But these two.h files can be found by following the path given.

ZbyLGsc, 2020-03-14:
No, we do not have any of such files needed. It must be caused by some of your other projects.

closed 2020-03-14 by ZbyLGsc

Quixote-Sin, 2020-03-15:
I haven't done other projects. I still don't know how such a problem happened. Could you please help me to see how to solve it.

`In file included from /home/quixote-yang/catkin_ws/src/Firmware/src/lib/ecl/attitude_fw/ecl_pitch_controller.h:54:0,
                 from /home/quixote-yang/catkin_ws/src/Firmware/src/modules/fw_att_control/FixedwingAttitudeControl.hpp:36,
                 from /home/quixote-yang/catkin_ws/src/Firmware/src/modules/fw_att_control/FixedwingAttitudeControl.cpp:34:
/home/quixote-yang/catkin_ws/src/Firmware/src/lib/ecl/attitude_fw/ecl_controller.h:51:17: fatal error: ecl.h: 没有那个文件或目录
compilation terminated.
Firmware/src/modules/fw_att_control/CMakeFiles/modules__fw_att_control.dir/build.make:62: recipe for target 'Firmware/src/modules/fw_att_control/CMakeFiles/modules__fw_att_control.dir/FixedwingAttitudeControl.cpp.o' failed
make[2]: *** [Firmware/src/modules/fw_att_control/CMakeFiles/modules__fw_att_control.dir/FixedwingAttitudeControl.cpp.o] Error 1
CMakeFiles/Makefile2:12818: recipe for target 'Firmware/src/modules/fw_att_control/CMakeFiles/modules__fw_att_control.dir/all' failed
make[1]: *** [Firmware/src/modules/fw_att_control/CMakeFiles/modules__fw_att_control.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
`

`In file included from /home/quixote-yang/catkin_ws/src/Firmware/src/modules/fw_pos_control_l1/FixedwingPositionControl.hpp:58:0,
                 from /home/quixote-yang/catkin_ws/src/Firmware/src/modules/fw_pos_control_l1/FixedwingPositionControl.cpp:34:
/home/quixote-yang/catkin_ws/src/Firmware/src/lib/ecl/l1/ecl_l1_pos_controller.h:65:21: fatal error: geo/geo.h: 没有那个文件或目录
compilation terminated.
Firmware/src/modules/fw_pos_control_l1/CMakeFiles/modules__fw_pos_control_l1.dir/build.make:62: recipe for target 'Firmware/src/modules/fw_pos_control_l1/CMakeFiles/modules__fw_pos_control_l1.dir/FixedwingPositionControl.cpp.o' failed
make[2]: *** [Firmware/src/modules/fw_pos_control_l1/CMakeFiles/modules__fw_pos_control_l1.dir/FixedwingPositionControl.cpp.o] Error 1
CMakeFiles/Makefile2:12881: recipe for target 'Firmware/src/modules/fw_pos_control_l1/CMakeFiles/modules__fw_pos_control_l1.dir/all' failed
make[1]: *** [Firmware/src/modules/fw_pos_control_l1/CMakeFiles/modules__fw_pos_control_l1.dir/all] Error 2
`

Quixote-Sin, 2020-03-15:
> No, we do not have any of such files needed. It must be caused by some of your other projects.

I haven't done other projects. I still don't know how such a problem happened. Could you please help me to see how to solve it.

In file included from /home/quixote-yang/catkin_ws/src/Firmware/src/lib/ecl/attitude_fw/ecl_pitch_controller.h:54:0, from /home/quixote-yang/catkin_ws/src/Firmware/src/modules/fw_att_control/FixedwingAttitudeControl.hpp:36, from /home/quixote-yang/catkin_ws/src/Firmware/src/modules/fw_att_control/FixedwingAttitudeControl.cpp:34: /home/quixote-yang/catkin_ws/src/Firmware/src/lib/ecl/attitude_fw/ecl_controller.h:51:17: fatal error: ecl.h: 没有那个文件或目录 compilation terminated. Firmware/src/modules/fw_att_control/CMakeFiles/modules__fw_att_control.dir/build.make:62: recipe for target 'Firmware/src/modules/fw_att_control/CMakeFiles/modules__fw_att_control.dir/FixedwingAttitudeControl.cpp.o' failed make[2]: *** [Firmware/src/modules/fw_att_control/CMakeFiles/modules__fw_att_control.dir/FixedwingAttitudeControl.cpp.o] Error 1 CMakeFiles/Makefile2:12818: recipe for target 'Firmware/src/modules/fw_att_control/CMakeFiles/modules__fw_att_control.dir/all' failed make[1]: *** [Firmware/src/modules/fw_att_control/CMakeFiles/modules__fw_att_control.dir/all] Error 2 make[1]: *** 正在等待未完成的任务....

In file included from /home/quixote-yang/catkin_ws/src/Firmware/src/modules/fw_pos_control_l1/FixedwingPositionControl.hpp:58:0, from /home/quixote-yang/catkin_ws/src/Firmware/src/modules/fw_pos_control_l1/FixedwingPositionControl.cpp:34: /home/quixote-yang/catkin_ws/src/Firmware/src/lib/ecl/l1/ecl_l1_pos_controller.h:65:21: fatal error: geo/geo.h: 没有那个文件或目录 compilation terminated. Firmware/src/modules/fw_pos_control_l1/CMakeFiles/modules__fw_pos_control_l1.dir/build.make:62: recipe for target 'Firmware/src/modules/fw_pos_control_l1/CMakeFiles/modules__fw_pos_control_l1.dir/FixedwingPositionControl.cpp.o' failed make[2]: *** [Firmware/src/modules/fw_pos_control_l1/CMakeFiles/modules__fw_pos_control_l1.dir/FixedwingPositionControl.cpp.o] Error 1 CMakeFiles/Makefile2:12881: recipe for target 'Firmware/src/modules/fw_pos_control_l1/CMakeFiles/modules__fw_pos_control_l1.dir/all' failed make[1]: *** [Firmware/src/modules/fw_pos_control_l1/CMakeFiles/modules__fw_pos_control_l1.dir/all] Error 2

ZbyLGsc, 2020-03-16:
@Quixote-Sin According to the outputs you provide, it is clear that you have a package named "Firmware" which cause the error. It has nothing to do with Fast-Planner. I suggest you to create a new workspace and do the compilation to avoid any problems caused by other packages.

Quixote-Sin, 2020-03-16:
Thank you very much for your patient suggestion! The problem is solved.