HKUST-Aerial-Robotics/Fast-Planner #18: Error during Compilation: cannot find -lpose_utils

mzahana, 2020-04-17:
Hi.
I am trying to compile this package using `catkin build`. Environment is Ubuntu 18 with ROS melodic. I am getting the following error. Could you please give some hints on how to solve this?
Thanks

```sh
Errors     << so3_disturbance_generator:make /home/aaal/catkin_ws/logs/so3_disturbance_generator/build.make.000.log                                                                                                                   
/usr/bin/ld: cannot find -lpose_utils
collect2: error: ld returned 1 exit status
make[2]: *** [/home/aaal/catkin_ws/devel/lib/so3_disturbance_generator/so3_disturbance_generator] Error 1
make[1]: *** [CMakeFiles/so3_disturbance_generator.dir/all] Error 2
make: *** [all] Error 2

```

ZbyLGsc, 2020-04-18:
You may try compiling the pose_utils packange in the Utils folder, before compiling this so3_disturbance_generator.

mzahana, 2020-04-19:
I tried
```sh
catkin build pose_utils
```
and
```sh
cd path/to/pose_utils
mkdir build && cd build
cmake .. && make && sudo make install
cd ~/catkin_ws
catkin build
```

I am still getting the same error
```sh
Errors     << so3_disturbance_generator:make /home/aaal/catkin_ws/logs/so3_disturbance_generator/build.make.003.log
/usr/bin/ld: cannot find -lpose_utils
collect2: error: ld returned 1 exit status
```

ZbyLGsc, 2020-04-20:
This could happen because we have not tested the package on melodic.
Someone seems to have fixed this problem in the pull request, and you may refer to https://github.com/HKUST-Aerial-Robotics/Fast-Planner/pull/6/https://github.com/HKUST-Aerial-Robotics/Fast-Planner/pull/6/.
Basically it change some linking target related to pose_utils in CMakeLists

ZbyLGsc, 2020-04-20:
@mzahana Another possible reason is that you use catkin build instead of catkin_make. According to the document of catkin_tool, the catkin build command builds each package in a workspace’s source space in **isolation** in order to prevent build-time cross-talk. This however can cause linking problem between the packages. I STRONGLY recommend you to create a new workspace and use catkin_make.

closed 2020-04-28 by ZbyLGsc

patrickpoirier51, 2020-04-30:
Hello @mzahana  :-)

The answers are in the PR 
https://github.com/HKUST-Aerial-Robotics/Fast-Planner/pulls

mzahana, 2020-05-01:
@ZbyLGsc thanks for the suggestions, the pr you mentioned worked for me. 

@patrickpoirier51 nice to meet you again :) and yes that PR compiled with me.