HKUST-Aerial-Robotics/Fast-Planner #20: Does the '3D Nav Goal' button works?

infinity0323, 2020-04-29:


infinity0323, 2020-04-29:
I changed the code in 'plan_manage/src/topo_replan_fsm.cpp' to the following:

` if (target_type_ == TARGET_TYPE::MANUAL_TARGET) {
      target_point_(0) = msg->poses[0].pose.position.x;
      target_point_(1) = msg->poses[0].pose.position.y;
      target_point_(2)=msg->poses[0].pose.position.z;
      std::cout << "manual: " << target_point_.transpose() << std::endl;
`

And I want to change the flying height when I run the 'kino_replan.launch'. Can you tell me how to do that？ Thanks a lot!

ZbyLGsc, 2020-04-29:
No currently.

In the previous version we did use the 3D Nav Goal, which is a customized RVIZ plugin to select 3D targets. However I found that its usage is complex and often confused users. Therefore it has been removed completely.

Now the only way to set 3D goals is to specify them in the .launch files.

infinity0323, 2020-04-30:
So that's it! If possible, could you send me the old version? If not, forget it. Thank you!
My email: helloword2336@126.com

infinity0323, 2020-04-30:
Anyway, thanks for sharing the current code. It's amazing! :)

linzgood, 2020-05-02:
could you send me the 3D Nav Goal version? My email: linz_good@126.com

ZbyLGsc, 2020-05-05:
@infinity0323 @linzgood You may check the previous version in the commit.

ZbyLGsc, 2020-05-05:
For example, use this commit https://github.com/HKUST-Aerial-Robotics/Fast-Planner/blob/cd8d28ef2e716063585b09687802ad940c9f7bda.

ZbyLGsc, 2020-05-05:
For example check this commit: https://github.com/HKUST-Aerial-Robotics/Fast-Planner/tree/cd8d28ef2e716063585b09687802ad940c9f7bda

infinity0323, 2020-05-06:
> For example check this commit: https://github.com/HKUST-Aerial-Robotics/Fast-Planner/tree/cd8d28ef2e716063585b09687802ad940c9f7bda

Thank you very much!

closed 2020-05-19 by ZbyLGsc