HKUST-Aerial-Robotics/Fast-Planner #24: Can this great project be used for 2D navigation of mobile robots?

shaoxiang, 2020-06-03:
Thank you very much for this great open source project. 
I want to use it in 2D navigation of mobile robot. Do you think it is suitable? Any suggestions？Thanks.

ZbyLGsc, 2020-06-06:
@shaoxiang Yes of course, 2D is even easier than 3D. Just set z  to some fixed values.

closed 2020-06-06 by ZbyLGsc

fxprime, 2020-06-08:
@ZbyLGsc Can i apply laser scan as sensor source (maybe convert to pointcloud first) for 2D navigation? Will it have some incompetible with algorithm? Thank you advanced.

ZbyLGsc, 2020-06-15:
@fxprime Yes of course, we can also take sensor pose and pointcloud as input to update the map.
You may check the .launch file for details.

LeeXujie, 2020-11-16:
> @shaoxiang Yes of course, 2D is even easier than 3D. Just set z to some fixed values.

Hello, Dr. Zhou. Which z should be set to a fixed value? And where to set it? Thank you very much!

LiJing9876, 2021-04-11:
Hey,
First of all, great repo!

Similar question here, my problem is that the kino algorithm is planning a path that is not in the same direction as yaw direction, i.e in the case of a ground robot moving sidewise...
How can I overcome this issue?
Thanks ahead!

zrd1234, 2022-09-16:
> @shaoxiang是的，当然，2D甚至比3D更容易。只需将 z 设置为一些固定值即可。
Hello, can you tell me how to modify it? I want to make a 2D plan. Thank you.

LinSC666, 2023-04-07:
> @shaoxiang Yes of course, 2D is even easier than 3D. Just set z to some fixed values.
Hello,may I ask if you have successfully run the fast planner on the 2D differential-wheel robot?I found that changing the search state space to (v, w) resulted in a particularly strange trajectory for the front-end output dynamics constraint. Overall, it was not smooth and often had broken lines. Because I have just read Fast Planner's paper and code, and I still don't understand many of its essence, such as if after changing to a differential car, do UAV and differential robots use the Pontryagin principle to solve two-point boundary value problems in the same way?I transplanted this code intact because I don't know how to change the coefficients of many matrices，it's really just set z to some fixed calues? 
Looking forward to receiving feedback and guidance, thank you!

LinSC666, 2023-04-07:
> 
Hello,may I ask if you have successfully run the fast planner on the 2D differential-wheel robot?I found that changing the search state space to (v, w) resulted in a particularly strange trajectory for the front-end output dynamics constraint. Overall, it was not smooth and often had broken lines. Because I have just read Fast Planner's paper and code, and I still don't understand many of its essence, such as if after changing to a differential car, do UAV and differential robots use the Pontryagin principle to solve two-point boundary value problems in the same way?I transplanted this code intact because I don't know how to change the coefficients of many matrices，it's really just set z to some fixed calues?
Looking forward to receiving feedback and guidance, thank you!