HKUST-Aerial-Robotics/Fast-Planner #25: questions about FastPlannerManager::planYaw()

WangChanglong17, 2020-06-07:
Hi, I have some questions about the yaw plan function.
1. How is this derived？
  states2pts << 1.0, -dt_yaw, (1 / 3.0) * dt_yaw * dt_yaw, 1.0, 0.0, -(1 / 6.0) * dt_yaw * dt_yaw, 1.0,
      dt_yaw, (1 / 3.0) * dt_yaw * dt_yaw;
2. the size of control points is  seg_num + 3, why ?  I find that In calcWaypointsCost function there are only seg_num+2 control points being used to calc cost. Can I undestand that the extra point is used to contrain end_yaw? But I dont know how (states2pts * end_yaw and states2pts * start_yaw) works, as mentioned in ques 1.
Thanks!

ZbyLGsc, 2020-06-07:
These are basic mathematic of B-spline.
For a 3-degree B-spline with N segments, there are N+3 control points, which answers your second questions.

As for the the first question, for a 3-degree B-spline the start and end state (0,1,2 order derivatives) are uniquely determined by the first and last 3 control points. The matrix state2pts is the relation between the state and control points.
You may check the paper "General matrix representations for b-splines" for the derivation.

ZbyLGsc, 2020-06-07:
@WangChanglong17

WangChanglong17, 2020-06-08:
I know that the start and end state are uniquely determined by the first and last 3 control points. So that yaw.block(seg_num, 0, 3, 1) = states2pts * start_yaw is an initialization? But after being optimized,  
this initialization seems meaningless, because 1/6(yaw(0) + 4yaw(1) + yaw(2)) should be constrained to waypoints_(0) which is not equal to start_yaw. this is confusing.

ZbyLGsc, 2020-06-09:
No, the first and last 3 points remain fixed throughout the optimization, so the boundary state will not be changed.

ZbyLGsc, 2020-06-09:
Actually the first and last waypoint constraints take no effect. 
In the code I do add them into the cost function, just for convenience of implementation.
But since the first and last 3 control points are already fixed, the optimization will not change them.
The associated cost of waypoints_(0) remains a constant term.

WangChanglong17, 2020-06-09:
thanks very much for your reply. Now I understand. Great work!

closed 2020-06-09 by WangChanglong17