HKUST-Aerial-Robotics/Fast-Planner #28: Which frame is considered for depth or cloud topic?

mzahana, 2020-06-21:
Hi @ZbyLGsc ,
Do depth image or cloud messages have to be in particular frame?
Usually they are in the sensor frame.
Is there any required transformation?

Thanks.

ZbyLGsc, 2020-06-22:
You should specify the pose of the camera frame in the world frame, if you use depth.
This would be used to transform the projected points to the world frame.
Similiar for cloud. Please check the comment of .launch files

mzahana, 2020-06-22:
@ZbyLGsc  you mean the [camera_pose_tpoic](https://github.com/HKUST-Aerial-Robotics/Fast-Planner/blob/master/fast_planner/plan_manage/launch/topo_replan.launch#L20)  ?
And is it used for both depth and point cloud topics?
Thanks.

mzahana, 2020-06-22:
It is mentioned in [the launch](https://github.com/HKUST-Aerial-Robotics/Fast-Planner/blob/master/fast_planner/plan_manage/launch/topo_replan.launch#L24) file not to set the camera pose argument if cloud topic is used. In this case how is the point cloud transformed? Or it has to be transformed to world frame before it's passed to the planner?
```
<!-- don't set camera pose and depth, if you already set this one! -->
<arg name="cloud_topic" value="/pcl_render_node/cloud"/>
```

ZbyLGsc, 2020-06-22:
Yes you are right. The cloud interface is not well explained, I will fix this problem later. Thanks for reporting it.

closed 2020-07-01 by ZbyLGsc