HKUST-Aerial-Robotics/Fast-Planner #30: Avoiding large obstacles

mzahana, 2020-06-23:
Hi @ZbyLGsc 
I am trying to use topo_replan in Gazebo simulation setup where I have some large obstacles like building and warehouses. With the default parameters of topo_replan, the drone doesn't find a good path around, e.g. large building (see attached picture) and ends up crashing into it. I would like to know what parameters can be tuned to be able to avoid such large obstacles (not necessarily flat objects).

Thanks
[image]

ZbyLGsc, 2020-06-24:
@mzahana The released topo_replan is not integrated with any global path planner such as A* / RRT*.
Currently it simply uses the straight line connecting the start and end point as the global path, and replans around the straight line. You may change <param name="topo_prm/sample_inflate_y" value="3.5" type="double"/> to enlarge the area of replanning region but it may not work for very large obstacles.
You can also try the kino_replan, which does not have such problem since a global path finder is used.

Could you please star this repo if it helps your project, and close the solved issues? Thanks. :)

mzahana, 2020-06-24:
@ZbyLGsc Thank you for the clarification. I found an issue related to the `/sdf_map/occupancy_inflate` and it might be part of the issue here. Sometimes, when the drone is flying to a waypoint, the `/sdf_map/occupancy_inflate` becomes empty some where in on the way although there are obstacles in front of the drone, see picture below. Is this an issue? If yes what could be the cause?
[image]

mzahana, 2020-06-24:
@ZbyLGsc Just to do more debugging, I noticed that the cloud in `/sdf_map/occupancy_inflate` is not well aligned with the actual depth cloud, see pictures below. Would this be related to camera intrinsic parameters, or something else ?

[image]

[image]

mzahana, 2020-06-24:
The alignment issue is solved after setting the right camera intrinsic params. The issue of having empty `/sdf_map/occupancy_inflate` is still there even if there are obstacles in front of the drone.

ZbyLGsc, 2020-06-24:
@mzahana I think this can be solved by changing the maximum map size: map_size_x/map_size_y/map_size_z. By default they are 40, 40, 5.

ZbyLGsc, 2020-06-24:
Please be aware of the memory usage. The mapping is not optimized for very large scene :). You may refer to https://github.com/HKUST-Aerial-Robotics/Fast-Planner/issues/13.

mzahana, 2020-06-24:
@ZbyLGsc I increased the map size already. However, I suspect the map size is the issue as obstacles can be just couple of meters away and in the FOV, but still not seen in the `/sdf_map/occupancy_inflate`

ZbyLGsc, 2020-06-24:
@mzahana Based on the image you provide I think there cloud not be other issue.
The obstacles in the FOV is about 40 meters away from the origin (0,0,0), so the map_size_x should be at least 80 (from -40 to 40 meters), otherwise the sdf_map will not map them.
Give me more details if the problem is not solved yet.

mzahana, 2020-06-25:
@ZbyLGsc Problem solved. Obstacles were outside the defined map size and, hence, not included in the planning. Thanks for your feedback.

closed 2020-06-25 by mzahana