HKUST-Aerial-Robotics/Fast-Planner #35: want to save more maps

skyaid, 2020-07-16:
The default param only saves a little map and disappears quickly. If you increase the deth scaling factor, does it store more maps?

ZbyLGsc, 2020-07-16:
@skyaid There are two things you should be aware of: (1) the map only keeps data within an area around the current FOV. (2) only part of the kept data is visualized in RVIZ (to reduce the burden of visualization).

Therefore if you want to see more map, you should: (1) increase the area of local map, by setting a larger 'sdf_map/local_map_margin'. (2) In *visCallback*, set *publishMapInflate(true)*, which will display all map data in RVIZ;

lgh5054, 2020-07-17:
@ZbyLGsc  Thank you for your quick answer.

As you said, it's working normally after modifying it.

initially 
Why do I set
                    map x 4
                    map y 4
                    map z 4?

Can I remove the function when I try to use it outdoors?

closed 2020-07-18 by ZbyLGsc