HKUST-Aerial-Robotics/Fast-Planner #36: segmentation fault error

Lee0326, 2020-07-19:
The change in the pr #4 has been done to build it under Ubuntu 18.04. The segmentation fault error occurs when I set the 2D nav goal in rviz:

> Stack trace (most recent call last):
#4    Object "[0x7ffe7d4eb93f]", at 0x7ffe7d4eb93f, in 
#3    Object "/home/colin/Repos/catkin_fast_planner/devel/lib/libpath_searching.so", at 0x7ff6fd00cf82, in 
#2    Object "/lib/x86_64-linux-gnu/libgcc_s.so.1", at 0x7ff6fab22d06, in _Unwind_Resume
#1    Object "/lib/x86_64-linux-gnu/libgcc_s.so.1", at 0x7ff6fab223e5, in 
#0    Object "/lib/x86_64-linux-gnu/libgcc_s.so.1", at 0x7ff6fab21f34, in 
Segmentation fault (Signal sent by the kernel [(nil)])
[fast_planner_node-1] process has died [pid 19276, exit code -11, cmd /home/colin/Repos/catkin_fast_planner/devel/lib/plan_manage/fast_planner_node /odom_world:=/state_ukf/odom /sdf_map/odom:=/state_ukf/odom /sdf_map/cloud:=/pcl_render_node/cloud /sdf_map/pose:=/pcl_render_node/camera_pose /sdf_map/depth:=/pcl_render_node/depth __name:=fast_planner_node __log:=/home/colin/.ros/log/5440e0a2-c9e1-11ea-980d-802bf9e83091/fast_planner_node-1.log].
log file: /home/colin/.ros/log/5440e0a2-c9e1-11ea-980d-802bf9e83091/fast_planner_node-1*.log

Does anyone has this problem? Thanks a lot

closed 2020-11-02 by ZbyLGsc

nikola-j, 2020-12-11:
Hi @Lee0326, I've solved this same problem by changing my gcc/g++ version from 8 to 7

Lee0326, 2021-01-01:
> Hi @Lee0326, I've solved this same problem by changing my gcc/g++ version from 8 to 7

It works! Thanks a lot

chengyh23, 2022-07-23:
> Hi @Lee0326, I've solved this same problem by changing my gcc/g++ version from 8 to 7

It works! (change g++/gcc version from 9 to 5)
Thank you very much. how did you find this solution?