HKUST-Aerial-Robotics/Fast-Planner #41: Eigen problem by running the demo topo_replan.launch

Daihn6, 2020-08-04:
When I run: roslaunch plan_manage topo_replan.launch, and give a target point, it appears an error that:

```
Stack trace (most recent call last):
#13   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#12   Object "/home/daihn666/catkin_ws/devel/lib/plan_manage/fast_planner_node", at 0x41c148, in _start
#11   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fb24fd2083f, in __libc_start_main
#10   Object "/home/daihn666/catkin_ws/devel/lib/plan_manage/fast_planner_node", at 0x4197db, in main
#9    Object "/opt/ros/kinetic/lib/libroscpp.so", at 0x7fb252688e9a, in ros::spin()
#8    Object "/opt/ros/kinetic/lib/libroscpp.so", at 0x7fb2526a3e38, in ros::SingleThreadedSpinner::spin(ros::CallbackQueue*)
#7    Object "/opt/ros/kinetic/lib/libroscpp.so", at 0x7fb25264723a, in ros::CallbackQueue::callAvailable(ros::WallDuration)
#6    Object "/opt/ros/kinetic/lib/libroscpp.so", at 0x7fb252645837, in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*)
#5    Object "/opt/ros/kinetic/lib/libroscpp.so", at 0x7fb25261f2dc, in ros::TimerManager<ros::Time, ros::Duration, ros::TimerEvent>::TimerQueueCallback::call()
#4    Object "/home/daihn666/catkin_ws/devel/lib/plan_manage/fast_planner_node", at 0x433927, in fast_planner::TopoReplanFSM::execFSMCallback(ros::TimerEvent const&)
#3    Object "/home/daihn666/catkin_ws/devel/lib/plan_manage/fast_planner_node", at 0x43343e, in fast_planner::TopoReplanFSM::callTopologicalTraj(int)
#2    Object "/home/daihn666/catkin_ws/devel/lib/plan_manage/fast_planner_node", at 0x43bdff, in fast_planner::FastPlannerManager::planGlobalTraj(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)
#1    Object "/home/daihn666/catkin_ws/devel/lib/libpoly_traj.so", at 0x7fb25233349e, in minSnapTraj(Eigen::Matrix<double, -1, -1, 0, -1, -1> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, Eigen::Matrix<double, -1, 1, 0, -1, 1> const&)
#0    Object "/home/daihn666/catkin_ws/devel/lib/libpath_searching.so", at 0x7fb253036dd0, in void Eigen::internal::call_assignment_no_alias<Eigen::Matrix<double, -1, 1, 0, -1, 1>, Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1, 0, -1, 1> >, Eigen::internal::assign_op<double> >(Eigen::Matrix<double, -1, 1, 0, -1, 1>&, Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1, 0, -1, 1> > const&, Eigen::internal::assign_op<double> const&)
Segmentation fault (Signal sent by the kernel [(nil)])
[fast_planner_node-1] process has died [pid 21125, exit code -11, cmd /home/daihn666/catkin_ws/devel/lib/plan_manage/fast_planner_node /odom_world:=/state_ukf/odom /sdf_map/odom:=/state_ukf/odom /sdf_map/cloud:=/pcl_render_node/cloud /sdf_map/pose:=/pcl_render_node/camera_pose /sdf_map/depth:=/pcl_render_node/depth __name:=fast_planner_node __log:=/home/daihn666/.ros/log/2003cfbe-d645-11ea-8617-485f9981cbdb/fast_planner_node-1.log].
log file: /home/daihn666/.ros/log/2003cfbe-d645-11ea-8617-485f9981cbdb/fast_planner_node-1*.log
```

I can't address this Eigen problem, and don't know how to solve it. My Eigen version is 3.2.92. My college has also this version Eigen, but he can run this demo without problem.

closed 2020-11-02 by ZbyLGsc

liuzhenboo, 2020-11-11:
have you soved it？

liuzhenboo, 2020-11-12:
in polynomial_traj.cpp replace
`    A.block(k * 6, k * 6, 6, 6) = Ab;
`
to
`    A.block<6, 6>(6 * k, 6 * k) = Ab;
`
at function minSnapTra()j

Daihn6, 2020-11-17:
> in polynomial_traj.cpp replace
> ` A.block(k * 6, k * 6, 6, 6) = Ab;`
> to
> ` A.block<6, 6>(6 * k, 6 * k) = Ab;`
> at function minSnapTra()j

Ok, Thank you!