HKUST-Aerial-Robotics/Fast-Planner #50: Where is heading_planner?

dssdyx, 2020-08-31:
I see there is a series of parameters named "heading_planner" in topo_algorithm.xml. But I can not change them to reduce the yaw rate and find where they are in codes. Could you tell me which these parameters mean ?

ZbyLGsc, 2020-09-01:
These parameters are not used in the current planner but will be used in a future release. Currently the yaw is controlled to face toward the flight direction (or name it velocity tracing yaw). If you want to reduce the yaw rate, try reduce parameter: "optimization/lambda7". This will relax the velocity tracking constraint and enable higher smoothness.

The code for yaw planning is in "void FastPlannerManager::planYaw(const Eigen::Vector3d& start_yaw)".

dssdyx, 2020-09-02:
@ZbyLGsc Thank you for your answer!

closed 2020-09-02 by dssdyx