HKUST-Aerial-Robotics/Fast-Planner #57: Importing Real environment

Amit10311, 2020-09-22:
Thank you for this great package for simulating MAV. I am running Ubuntu 16.04. The simulator is working perfectly. I want to set up my own environment. You have created the environment with the use of a` mockamap` package. 

1. I would like to know while flying in an outdoor environment how to provide the map to the simulator? 
2.  Can we provide the octomap data to the simulator? 

Can you please guide me on how to set up the environment in your package.

I look forward to hearing from you. Thank you for your valuable time.

ZbyLGsc, 2020-09-23:
We do not have any API to import a large environment directly. The input to the planner and mapping is either depth image or point cloud from sensors. You may check the launch file for setting the respective topics

closed 2020-09-25 by ZbyLGsc

Amit10311, 2020-10-06:
Dear Boyu Zhou,  I would like to know that the local Map generation for motion planning in [https://github.com/HKUST-Aerial-Robotics/FIESTA](url) is used to generate the local map in `Fast-Planner`?

ZbyLGsc, 2020-10-06:
Most key algorithms (except the ESDF computation) are the same, but the code is completely rewritten.