HKUST-Aerial-Robotics/Fast-Planner #58: Question about the smoothness Cost

taec1227, 2020-09-24:
Hi, I'm looking at the bspline optimization part in your project, and have a question about the calcSmoothnessCost function.

In the paper, the cost is formulated as: 
[image]

However, from the code, it shows like
jerk = q[i + 3] - 3 * q[i + 2] + 3 * q[i + 1] - q[i];
cost += jerk.squaredNorm();

Is the function in paper wrong? If so, what is the meaning of the code one?

ZbyLGsc, 2020-09-25:
1. Cost in the paper corresponds to second order derivative of B-spline, while that in the code corresponds to third order one (jerk). Both of them can be used.
2. There is no 'hard coded bound', we penalize points that are close to distance. The distance is queried from ESDF.

closed 2020-09-25 by ZbyLGsc

taec1227, 2020-09-29:
Thanks for your reply, and there is another function named 'evaluateDeBoor' which confused me a lot. I know that the deBoor is the recursion of the b-spline basis, however, the 'evaluateDeBoor' seems to use control points to do a recursion, I didn't find any theory to support it, do I misunderstand any useful formulation?

ZbyLGsc, 2020-09-29:
What you describe is called Cox-de Boor recursion formula, which evaluates the basis function explicitly. de Boor's algorithm is different. It is based on an equivalent recursive formula that does not evaluate the basis function.