HKUST-Aerial-Robotics/Fast-Planner #59: Build fails on Ubuntu 18.04.5 with ROS Melodic

pauljurczak, 2020-10-07:
Unfortunately, I don't see a reason why. Here are the last few lines of catkin output:
```
/home/paul/hkust/src/Fast-Planner/fast_planner/path_searching/src/astar.cpp: In member function ‘double fast_planner::Astar::getDiagHeu(Eigen::Vector3d, Eigen::Vector3d)’:
/home/paul/hkust/src/Fast-Planner/fast_planner/path_searching/src/astar.cpp:256:25: warning: ‘h’ may be used uninitialized in this function [-Wmaybe-uninitialized]
   return tie_breaker_ * h;
                         ^
[ 69%] Linking CXX shared library /home/paul/hkust/devel/lib/libpath_searching.so
[ 69%] Built target path_searching
Makefile:159: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```
I followed the instructions from https://github.com/HKUST-Aerial-Robotics/Fast-Planner#3-setup-and-config

zhangzhijie1998, 2020-10-14:
run "catkin_make --debug" to find more information. I encountered the same error. The reason might be related to libnlopt-dev.

xdke666, 2020-10-22:
ubuntun16.04 , same error, "Built target path_searching" failed

xdke666, 2020-10-22:
I have solved this problem. Generally, the. So file is missing. There are prompts during compilation. Install the missing library to the correct path

closed 2020-11-02 by ZbyLGsc

pushkar-khetrapal, 2020-11-24:
Hi, @xdke666 how did you found the solution? I'm getting the same error.