HKUST-Aerial-Robotics/Fast-Planner #68: Sending the UAV to multiple points without using 2D Nav Goal on rviz

bharat787, 2020-11-04:
I'm trying to make the UAV go on a set path. I changed the flight_type to 2 in kino_replan.launch to use and set the preset points in  kino_alogorithm.xml and kino_replan.launch.
But during execution I still have to use the 2D Nav Goal button in rviz to make the UAV navigate to the next preset point.

Is there a way to make it automatically go through all the set points.

Thanks

ZbyLGsc, 2020-11-06:
preset multiple points are not supported for kino_replan; use topo_replan instead.
Thanks for reporting the problem, I will correct the comment in launch file soon.

ZbyLGsc, 2020-11-06:
If you are determined to use kino-replan, then you should modify the state machine(KinoReplanFSM) to enable continuous execution of the waypoints without using 2D Nav Goal to trigger it repeatedly.

bharat787, 2020-11-06:
Thanks for the suggestion i'll try them out

bharat787, 2020-11-06:
> Presetting multiple points is not supported for kino_replan, the waypoints will be executed one by one;
> Use topo_replan instead in which you only need to trigger the planner once.

Facing the same issue on topo_replan.

ZbyLGsc, 2020-11-06:
Did you set flight_type=3 and set point_num and point0, point1 ... pointn properly?

bharat787, 2020-11-07:
> Did you set flight_type=3 and set point_num and point0, point1 ... pointn properly?

[image]

The comments did not mention there was a mode 3, but it works now!

Thanks for the fast responses.

closed 2020-11-13 by ZbyLGsc