HKUST-Aerial-Robotics/Fast-Planner #70: SDF map removes obstacles out of FOV

haolong12321, 2020-11-05:
Hi and thanks again for this great package.

I'm running the kino_replan using my own depth camera.
When obstacles get out of FOV they are being removed from SDF map and the algo "forgets" them.
Is there any way to extend the range that the algo is keeping SDF map data?

Thanks ahead.

ZbyLGsc, 2020-11-06:
Pls refer to this issue: https://github.com/HKUST-Aerial-Robotics/Fast-Planner/issues/60

LiJing9876, 2020-11-09:
Do you mean that the way to extend the range of the SDF map data is to comment out lines 645-708 in sdf_map.cpp (clearing data outside the local map)?

haolong12321, 2020-11-09:
Do you mean that the way to extend SDF map data range is to comment out lines 645-708 in sdf_map.cpp (clear data outside the local range)?

liuzhenboo, 2020-11-12:
> Hi and thanks again for this great package.
> 
> I'm running the kino_replan using my own depth camera.
> When obstacles get out of FOV they are being removed from SDF map and the algo "forgets" them.
> Is there any way to extend the range that the algo is keeping SDF map data?
> 
> Thanks ahead.

`    <param name="sdf_map/local_map_margin" value="500"/>
`
you can set the param bigger!

closed 2020-11-13 by ZbyLGsc

devkishan920, 2026-01-07:
> [@haolong12321](https://github.com/haolong12321) Hi, Hope you are fine i wanna know how you are adding your own depth camera what changes you made to run it.. thankyou hoping for your reply

@EhrazImam did you find any solution??