HKUST-Aerial-Robotics/Fast-Planner #71: Unable to change height

Stavya1993, 2020-11-06:
Using the kino_replan, I am using the /move_base_simple/goal to move the drone. Whenever, I change only the height of the system, the kino_replan fails to find trajectory. Additionally, it doesnt seem to change the height when x and y are also changed. I have changed the virtual cieling height to 10 in the kino_algorithm.launch file.

ZbyLGsc, 2020-11-06:
1. /move_base_simple/goal will set the height to a fixed value by default, since it is not convenient to specify the height with it.
2. There is a known issue: when the target x,y position is identical to the current x,y position the planner may not find paths.

ZbyLGsc, 2020-11-06:
Currently one can only specify the target height by the preset waypoints in the launch file.

Stavya1993, 2020-11-06:
Thanks. This makes sense!

closed 2020-11-06 by Stavya1993