HKUST-Aerial-Robotics/Fast-Planner #72: Abnormal Behaviour using kino_replan

Stavya1993, 2020-11-06:
[image]

As you can see, even though there is no obstacle in front, the algorithm performs an obstacle avoiding maneuver. Just wanted to confirm how I can fix this behaviour.

Also wanted to extend my kudos to developers of this amazing algorithm!

ZbyLGsc, 2020-11-06:
Have you change the max map size of the x,y,z axes? When your scene is too large the default map size may not be large enough to cover it. You may change the size in launch file.

Stavya1993, 2020-11-06:
I have, Ive set the map size to 100 in x and y and 10 in z . The command itself is to move the quad to y = 18, x= -9, z =2.

ZbyLGsc, 2020-11-06:
Then I would suggest again to check the mapping module. Although the in the simulator we see no obstacles, it seem the depth camera does see something and generate a 'wall' in front of it.

Stavya1993, 2020-11-06:
Sure, Ill  check that again, post it when i find something! Will close the issue with the post to help others if they are stuck with similar problems / or in a few days if I cant find anything. Thanks!

closed 2020-11-13 by ZbyLGsc