HKUST-Aerial-Robotics/Fast-Planner #75: lack of files for header

gcc18, 2020-11-10:
Some header are not exist  so the project can not be compiled correctly

> 
fatal error: multi_map_server/MultiOccupancyGrid.h: No such file or directory
 #include <multi_map_server/MultiOccupancyGrid.h>
> 

> 
fatal error: plan_manage/Bspline.h: No such file or directory
 #include <plan_manage/Bspline.h>
>

closed 2020-12-19 by gcc18

mehhdiii, 2021-12-14:
Did the issue resolve? I seem to be having the same issue. I am on Ubuntu 18.04 (ROS melodic)

Tiga002, 2023-03-07:
@mehhdiii  @eliabntt  
BSpline.h is the header file for including the custom message type in msg/BSpline.msg

doche, 2023-07-22:
Did the issue resolve? I seem to be having the same issue. I am on Ubuntu 18.04 (ROS melodic),pls help!

doche, 2023-07-22:
> @mehhdiii @eliabntt BSpline.h is the header file for including the custom message type in msg/BSpline.msg

so? how to solve it? thanks