HKUST-Aerial-Robotics/Fast-Planner #79: About max_vel and max_acc parameters

vislero, 2020-11-19:
Hi. 
First of all, thank you for nice work sharing.

We have tested your mapping and path planning algorithms on our LiDAR(Ouster)  and LOAM.

[image]

But when we decrease 1/10 times max_vel, max_acc, searching path (Red line in below figures) are diverging (below figure).

[image]

And we found the similar problem in #64 .

So, we decrease 1/10 times search/resolution_astar in kino_algorithm.xml file like max_vel, max_acc.

But the result are not same as origin parameters. 

How to configure other parameters when we want to decrease max_vel and max_acc ?

Thank you.

ZbyLGsc, 2020-11-23:
@vislero After some checks we find it is not only a parameter setting problem, but also a bug caused by B-spline parameterization. We just fix the problem and in our test the planner works under max_vel= max_acc = 0.5. Besides the params you mentioned, we also suggest to set search/max_tau to a larger value. This param represents the duration of each motion primitive in the path searching.

vislero, 2020-12-03:
@ZbyLGsc 
Thank you very much !

closed 2020-12-03 by vislero