HKUST-Aerial-Robotics/Fast-Planner #81: Migrating to ubuntu 18.04/ROS melodic

realdealneil, 2020-11-26:
I tried building this in Ubuntu 18.04, and I'm running into a couple of issues.  I have fixed one, but not the second:

1) In uav_simulator/so3_control/src/so3_control_nodelet.cpp, PLUGINLIB_DECLARE_CLASS has been deprecated and replaced with PLUGINLIB_EXPORT_CLASS.  I fixed this and it now compiles.  
2) When re-running catkin_make, I get this error:

make[2]: *** No rule to make target '/usr/local/lib/libnlopt.so', needed by '/home/neil/devel/raptor_quad/raptor_ws/devel/lib/libbspline_opt.so'.  Stop.
CMakeFiles/Makefile2:3103: recipe for target 'Fast-Planner/fast_planner/bspline_opt/CMakeFiles/bspline_opt.dir/all' failed
make[1]: *** [Fast-Planner/fast_planner/bspline_opt/CMakeFiles/bspline_opt.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

I am trying to track this one down.  Any idea how to fix this?

realdealneil, 2020-11-26:
I see that others have addressed this in some recent pull requests.  I'll try following what they've done.

ZbyLGsc, 2020-12-10:
The project has been tested comprehensively on Ubuntu 18.04. Pls follow the new doc in readme to deploy it.

zwn, 2020-12-10:
@ZbyLGsc What a great news! Now I can really simplify my docker build and remove the patch I have there: https://github.com/zwn/robotic-dockers/blob/master/fast-planner/fp.patch

ZbyLGsc, 2020-12-11:
@zwn Good to hear!