HKUST-Aerial-Robotics/Fast-Planner #84: catkin_make ERROR

cuizhi, 2020-12-16:
Ubuntu18.04+PCL1.7+OpenCV3.4.13
 
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/ustb/FUEL_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/ustb/Prometheus/devel;/opt/ros/melodic
-- This workspace overlays: /home/ustb/Prometheus/devel;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/ustb/FUEL_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
`

`
[100%] Linking CXX executable /home/ustb/FUEL_ws/devel/lib/plan_manage/fast_planner_node
CMakeFiles/fast_planner_node.dir/src/planner_manager.cpp.o: In function `fast_planner::FastPlannerManager::initPlanModules(ros::NodeHandle&)':
/home/ustb/FUEL_ws/src/Fast-Planner/fast_planner/plan_env/include/plan_env/sdf_map.h:115: undefined reference to `cv::Mat::Mat()'
/home/ustb/FUEL_ws/src/Fast-Planner/fast_planner/plan_env/include/plan_env/sdf_map.h:115: undefined reference to `cv::Mat::Mat()'
collect2: error: ld returned 1 exit status
Fast-Planner/fast_planner/plan_manage/CMakeFiles/fast_planner_node.dir/build.make:448: recipe for target '/home/ustb/FUEL_ws/devel/lib/plan_manage/fast_planner_node' failed
make[2]: *** [/home/ustb/FUEL_ws/devel/lib/plan_manage/fast_planner_node] Error 1
CMakeFiles/Makefile2:6078: recipe for target 'Fast-Planner/fast_planner/plan_manage/CMakeFiles/fast_planner_node.dir/all' failed
make[1]: *** [Fast-Planner/fast_planner/plan_manage/CMakeFiles/fast_planner_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed
`

cuizhi, 2020-12-23:
Version clash of Opencv.

closed 2020-12-23 by cuizhi

Jackyxtt, 2021-09-16:
Hi, I met the same problem. Could you please explain more specifically？

AndyYan2, 2022-03-25:
> Hi, I met the same problem. Could you please explain more specifically？

I meet the same. I replace OpenCV3.4.13 with OpenCV3.2.0. It works!