HKUST-Aerial-Robotics/Fast-Planner #85: Migrating to ubuntu 20.04/noetic

dulex123, 2020-12-16:
I am trying to build master branch in Noetic, apart from installing ros-noetic-pcl* and changing std=c++11 flags to std=c++14 in all cmake files (PCL 1.10 requires it) everything compiles. 

I **can run both** roslaunch plan_manage rviz.launch and roslaunch plan_manage kino_replan.launch

**But when I issue navgoal it crashes without much information:**

[fast_planner_node-1] process has died [pid 23197, exit code -11, cmd ~/catkin_ws/devel/lib/plan_manage/fast_planner_node /odom_world:=/state_ukf/odom /sdf_map/odom:=/state_ukf/odom /sdf_map/cloud:=/pcl_render_node/cloud /sdf_map/pose:=/pcl_render_node/camera_pose /sdf_map/depth:=/pcl_render_node/depth __name:=fast_planner_node __log:=~/.ros/log/c518fd5c-3f9b-11eb-9def-dd79eedcd81f/fast_planner_node-1.log].
log file: ~/.ros/log/c518fd5c-3f9b-11eb-9def-dd79eedcd81f/fast_planner_node-1*.log

**I have ran the second command with --log so I have output of fast_planner_node but it doesnt say much:**

hit: 0.619039
miss: -0.619039
min log: -1.99243
max: 2.19722
thresh log: 1.38629
origin_: -20 -10  -1
map size: 40 20  5
[FSM]: from INIT to WAIT_TARGET
[FSM]: state: WAIT_TARGET
[FSM]: state: WAIT_TARGET
[FSM]: state: WAIT_TARGET
[FSM]: state: WAIT_TARGET
[FSM]: state: WAIT_TARGET
[FSM]: state: WAIT_TARGET
[FSM]: state: WAIT_TARGET
[FSM]: state: WAIT_TARGET
[FSM]: state: WAIT_TARGET
[FSM]: state: WAIT_TARGET
[FSM]: state: WAIT_TARGET
[FSM]: state: WAIT_TARGET
[FSM]: state: WAIT_TARGET
[FSM]: state: WAIT_TARGET
Triggered!
[TRIG]: from WAIT_TARGET to GEN_NEW_TRAJ
[kino replan]: -----------------------
start:         -19           0 1.51788e-18, 0 0 0, 0 0 0
goal:-9.50174 -1.04982        1, 0 0 0
open set empty, no path!
use node num: 1
iter num: 1
[kino replan]: kinodynamic search fail!
reach horizon
[kino replan]: retry search success.

**Rviz minor issue: Simulation-robot marker is not showing it says: status:error for frame /world fixed frame [world] does not exist**

WANGSSSSSSS, 2021-04-04:
i meet this error too, have you solve this problem yet?

yjhdhr, 2021-04-05:
hello, the same problem ...

yjhdhr, 2021-04-05:
> i meet this error too, have you solve this problem yet?

HI, do you know how to fix it?

surfii3z, 2021-04-30:
Same here, did anyone figure out the fix?

Junking1, 2021-05-07:
You can fix it with https://github.com/HKUST-Aerial-Robotics/Fast-Planner/pull/107
OR with  https://github.com/Junking1/Fast-Planner-for-ubuntu20.04

dulex123, 2021-05-17:
@Junking1 It doesnt work for me on 20.04. It builds successfully and when I give it 2d nav goal it crashes? Any ideas?

Junking1, 2021-05-17:
> @Junking1 It doesnt work for me on 20.04. It builds successfully and when I give it 2d nav goal it crashes? Any ideas?

Is it the same mistake as before?In fact, I suggest you download code from https://github.com/Junking1/Fast-Planner-for-ubuntu20.04  and try again.

dulex123, 2021-05-17:
@Junking1 There are no errors. Exactly on setting 2D nav goal fast_planner node crashes. You have no issues? It works as expected?

Junking1, 2021-05-17:
@dulex123 Yes, it works.Maybe you can output something in your nlopt library, making sure you're using the nlopt library you installed from the source code instead of the library from ros.

ddefuijk, 2021-05-25:
@dulex123 I'm having the same problem. Did you find a solution for this?

dulex123, 2021-05-25:
@ddefuijk nope. If you succeed please share the solution.

ddefuijk, 2021-05-25:
@dulex123 It seems I made a typo while sourcing the workspace's setup.bash. I doubt that you have the same issue but you can check that.

DeveloperFuloo, 2021-11-24:
crash in the same problem..  anyone fixed it?

iamrajee, 2021-11-25:
> crash in the same problem.. anyone fixed it?

Stuck with the same issue for a long time now. Eventually, I had to shift to the melodic. There it works without any issue!

iamrajee, 2022-03-21:
@eliabntt Your solution doesn't work for me. Has anybody else got it working?

eliabntt, 2022-03-21:
Hi, do you get an error? I did exactly what I stated and it started working. Be sure to completely clean the build files (`catkin clean`/ `rm build devel logs -rf`)  after you removed the nlopt packages. Try also to remove the packages, reboot and install nlopt from source.

iamrajee, 2022-03-21:
> catkin_make -DCMAKE_CXX_STANDARD=14

I tried that, but no luck :( Thanks anyway!

eliabntt, 2022-03-21:
I don't see segfault here.

qycqycqyc, 2023-03-01:
@eliabntt Thank you very much!