HKUST-Aerial-Robotics/Fast-Planner #86: Eigen problem while running topo_replan.launch

bjtulcy, 2020-12-23:
I can run kino_replan.launch successfully, but I cannot run topo_replan.launch with following error.

`Stack trace (most recent call last):
#21   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#20   Object "/home/htsys/LCY/fast_planner_ws/devel/lib/plan_manage/fast_planner_node", at 0x41b2a8, in _start
#19   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f62470f983f, in __libc_start_main
#18   Object "/home/htsys/LCY/fast_planner_ws/devel/lib/plan_manage/fast_planner_node", at 0x418ed9, in main
#17   Object "/opt/ros/kinetic/lib/libroscpp.so", at 0x7f6249a36a6a, in ros::spin()
#16   Object "/opt/ros/kinetic/lib/libroscpp.so", at 0x7f6249a51a18, in ros::SingleThreadedSpinner::spin(ros::CallbackQueue*)
#15   Object "/opt/ros/kinetic/lib/libroscpp.so", at 0x7f62499f513a, in ros::CallbackQueue::callAvailable(ros::WallDuration)
#14   Object "/opt/ros/kinetic/lib/libroscpp.so", at 0x7f62499f36ff, in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*)
#13   Object "/opt/ros/kinetic/lib/libroscpp.so", at 0x7f62499ceabc, in ros::TimerManager<ros::Time, ros::Duration, ros::TimerEvent>::TimerQueueCallback::call()
#12   Object "/home/htsys/LCY/fast_planner_ws/devel/lib/plan_manage/fast_planner_node", at 0x431d0d, in fast_planner::TopoReplanFSM::execFSMCallback(ros::TimerEvent const&)
#11   Object "/home/htsys/LCY/fast_planner_ws/devel/lib/plan_manage/fast_planner_node", at 0x43182e, in fast_planner::TopoReplanFSM::callTopologicalTraj(int)
#10   Object "/home/htsys/LCY/fast_planner_ws/devel/lib/plan_manage/fast_planner_node", at 0x43b5f4, in fast_planner::FastPlannerManager::planGlobalTraj(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)
#9    Object "/home/htsys/LCY/fast_planner_ws/devel/lib/libpoly_traj.so", at 0x7f62496ea68f, in minSnapTraj(Eigen::Matrix<double, -1, -1, 0, -1, -1> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, Eigen::Matrix<double, -1, 1, 0, -1, 1> const&)
#8    Object "/home/htsys/LCY/fast_planner_ws/devel/lib/libpoly_traj.so", at 0x7f62496fc9f9, in Eigen::PlainObjectBase<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::PlainObjectBase<Eigen::Product<Eigen::Product<Eigen::Product<Eigen::Product<Eigen::Matrix<double, -1, -1, 0, -1, -1>, Eigen::Inverse<Eigen::Transpose<Eigen::Matrix<double, -1, -1, 0, -1, -1> > >, 0>, Eigen::Matrix<double, -1, -1, 0, -1, -1>, 0>, Eigen::Inverse<Eigen::Matrix<double, -1, -1, 0, -1, -1> >, 0>, Eigen::Matrix<double, -1, -1, 0, -1, -1>, 0> >(Eigen::DenseBase<Eigen::Product<Eigen::Product<Eigen::Product<Eigen::Product<Eigen::Matrix<double, -1, -1, 0, -1, -1>, Eigen::Inverse<Eigen::Transpose<Eigen::Matrix<double, -1, -1, 0, -1, -1> > >, 0>, Eigen::Matrix<double, -1, -1, 0, -1, -1>, 0>, Eigen::Inverse<Eigen::Matrix<double, -1, -1, 0, -1, -1> >, 0>, Eigen::Matrix<double, -1, -1, 0, -1, -1>, 0> > const&)
#7    Object "/home/htsys/LCY/fast_planner_ws/devel/lib/libpoly_traj.so", at 0x7f62496fc207, in 
#6    Object "/home/htsys/LCY/fast_planner_ws/devel/lib/libpoly_traj.so", at 0x7f62496fba93, in 
#5    Object "/home/htsys/LCY/fast_planner_ws/devel/lib/libpoly_traj.so", at 0x7f62496fb080, in 
#4    Object "/home/htsys/LCY/fast_planner_ws/devel/lib/libpoly_traj.so", at 0x7f62496f9569, in Eigen::PartialPivLU<Eigen::Matrix<double, -1, -1, 1, -1, -1> >::PartialPivLU<Eigen::Matrix<double, -1, -1, 1, -1, -1> >(Eigen::EigenBase<Eigen::Matrix<double, -1, -1, 1, -1, -1> > const&)
#3    Object "/home/htsys/LCY/fast_planner_ws/devel/lib/libpoly_traj.so", at 0x7f62496f8ff3, in Eigen::PartialPivLU<Eigen::Matrix<double, -1, -1, 1, -1, -1> >::compute()
#2    Object "/home/htsys/LCY/fast_planner_ws/devel/lib/libpoly_traj.so", at 0x7f62496f8936, in Eigen::internal::partial_lu_impl<double, 1, int>::blocked_lu(long, long, double*, long, int*, int&, long)
#1    Object "/home/htsys/LCY/fast_planner_ws/devel/lib/libpath_searching.so", at 0x7f624a3dd414, in Eigen::internal::general_matrix_matrix_product<long, double, 0, false, double, 0, false, 0>::run(long, long, long, double const*, long, double const*, long, double*, long, double, Eigen::internal::level3_blocking<double, double>&, Eigen::internal::GemmParallelInfo<long>*)
#0    Object "/home/htsys/LCY/fast_planner_ws/devel/lib/libpath_searching.so", at 0x7f624a3d629b, in 
Segmentation fault (Address not mapped to object [0x10])
`
It's similar with [#41 ] , but the operation dosen't work in this error.

note2020, 2021-01-04:
My Eigen version is 3.3.7

jzhzhang, 2021-06-13:
@note2020  I meet the same problem as you.  And the #41 doesn't work.  Do you find the solution?