HKUST-Aerial-Robotics/Fast-Planner #88: trajectory through the obstacle

buenos-dan, 2020-12-25:
This issue may be as same as for issue #69.
I record a video as follows, and I wonder where goes wrong?

https://user-images.githubusercontent.com/36566377/103113611-7eb8a180-4696-11eb-817d-83b2ccb7f201.mp4

I modify the kino_replan.launch file  as follows:
1.  \<arg name="odom_topic" value="/mavros/local_position/odom" /\>
2. \<arg name="cloud_topic" value="/camera/depth/points"/\>
3. \<param name="sdf_map/frame_id"      value="base_link"/\>     //kino_algorithm.xml

Btw, the frame size is 640*480.
Thanks for your help.

ZbyLGsc, 2020-12-28:
It is the problem of coordinate system. As you can see, the obstacle moves when the drone flies forward, which is clearly problematic.

ZbyLGsc, 2020-12-28:
You should specify the topic of cloud in the world frame instead of the body frame.

iamrajee, 2022-06-08:
We are also facing a similar issue in SITL while trying out [beomsu7's](https://github.com/beomsu7/Fast-Planner) fork. Any resolution?

https://user-images.githubusercontent.com/25712145/172568365-0666c820-42fd-4b14-9fa5-bbdccc1a3d8c.mp4