HKUST-Aerial-Robotics/Fast-Planner #102: Experiment in real world

codespring-lh, 2021-04-21:
Hi,I want to do some experiment using Fast-Planner in real world,but I have met some problems.I used Vins-Fusion with D435i and I have modified kino_replan.lauch.But there is no map. 
Thanks for your reply.
[image]
[image]
[image]
[image]

codespring-lh, 2021-04-21:
And here is my tf_tree.
[image]

sbdjyl, 2021-05-18:
I have the same problem. I used D435i and vins-mono, modify the source of the odometer and depth image in the launch file, but in the rviz that fastplaner opened, I could not see the environment of camera acquisition.

Have you solved your problem?

sbdjyl, 2021-05-20:
i have solved the problem.
if you know Chinese，you can see my blog https://blog.csdn.net/weixin_44413712/article/details/117065832