HKUST-Aerial-Robotics/Fast-Planner #103: Using Fast-Planner for 2D Robot

LiJing9876, 2021-04-25:
Hello,

First of all thanks again for this great repo, I've been working with it for a few months now and it yields great results.
Until now I've worked with a quadcopter and now I want to start working with a 2D robot (such as turtlebot).
I read the following issue but still have some questions...
[https://github.com/HKUST-Aerial-Robotics/Fast-Planner/issues/24]

My problem is that the kino algorithm is planning a path that is not necessarily in the same direction as the yaw direction, i.e in the case of a 2D ground robot the robot could not follow that path because it means moving sidewise...

How can I overcome this issue?
Thanks ahead!

zrd1234, 2022-09-16:
> Hello,
> 
> First of all thanks again for this great repo, I've been working with it for a few months now and it yields great results. Until now I've worked with a quadcopter and now I want to start working with a 2D robot (such as turtlebot). I read the following issue but still have some questions... [https://github.com/[/issues/24](https://github.com/HKUST-Aerial-Robotics/Fast-Planner/issues/24)]
> 
> My problem is that the kino algorithm is planning a path that is not necessarily in the same direction as the yaw direction, i.e in the case of a 2D ground robot the robot could not follow that path because it means moving sidewise...
> 
> How can I overcome this issue? Thanks ahead!
Hello, I am also studying 2d recently. Do you know how to modify it. Thank you.

LinSC666, 2023-04-07:
As a rookie, I also want to learn how to change to 2D space planning, especially non-holonomic robot.
[https://github.com/https://github.com/HKUST-Aerial-Robotics/Fast-Planner/issues/24]

LinSC666, 2023-04-07:
> As a rookie, I also want to learn how to change to 2D space planning, especially non-holonomic robot. [https://github.com/https://github.com/[/issues/24](https://github.com/HKUST-Aerial-Robotics/Fast-Planner/issues/24)]
Hello,may I ask if you have successfully run the fast planner on the 2D differential-wheel robot?I found that changing the search state space to (v, w) resulted in a particularly strange trajectory for the front-end output dynamics constraint. Overall, it was not smooth and often had broken lines. Because I have just read Fast Planner's paper and code, and I still don't understand many of its essence, such as if after changing to a differential car, do UAV and differential robots use the Pontryagin principle to solve two-point boundary value problems in the same way?I transplanted this code intact because I don't know how to change the coefficients of many matrices，it's really just set z to some fixed calues?
Looking forward to receiving feedback and guidance, thank you!

devkishan920, 2026-01-07:
> Hello,
>
> First of all thanks again for this great repo, I've been working with it for a few months now and it yields great results. Until now I've worked with a quadcopter and now I want to start [...]

@LiJing9876 , have you implemented the fast planner on real drone. Any help would be greatly appreciated.

LinSC666, 2026-01-07:
您好，邮件已收到，我会尽快回复您，感谢喔！Hello, the E-mail has been received, I will reply you as soon as possible, thank you!