HKUST-Aerial-Robotics/Fast-Planner #104: Questions about the time allocation

taec1227, 2021-04-26:
I looked at the source code about both kino and topo method. For the kino method, you keep calling the reallocateTime() function until it is feasible, however, for topo method, you just do the time allocation once with ratio. My question is can it ensure the safety when do it only once in topo planner?

ZbyLGsc, 2021-04-26:
A good question. When doing only one reallocation, the dynamic constraints (maximal vel., acc.) may be violated slightly, but since the designated dynamic limits is much smaller than the actual physical limits of the quadrotor, but violation is acceptable in practice. Also, lengthening the time of a trajectory many times can lead to over-conservative time allocation, which is harmful for high-speed flight.

taec1227, 2021-04-29:
> A good question. When doing only one reallocation, the dynamic constraints (maximal vel., acc.) may be violated slightly, but since the designated dynamic limits is much smaller than the actual physical limits of the quadrotor, the violation is acceptable in practice. Moreover, lengthening the time of a trajectory many times can lead to over-conservative time allocation, which is harmful for high-speed flight.

Thanks for your reply.
1. An average reallocation is harmful, so, why not just increase part of the time duration where vel, acc exceeds the maximum limitation? Because of the trajectory shape will change when part change is applied? 
2. I tested the b-spline optimization part with some topological paths, unluckily found that the final trajectory didn't avoid trajectory successfully, is it also the drawback or do I miss any important part?

closed 2021-07-21 by taec1227