HKUST-Aerial-Robotics/Fast-Planner #108: What flight controller I need to use?

FPSychotic, 2021-05-15:
Hi, what flight controller as pix4, ardupilot, betaflight, openpilot open should I use? and how comunicate with it (i.e. mavros, rosflight, msp ...).
would be very helpful to know the basic setup, nothing specific, just to make me a general idea of what knd of drone is used, as protocol and camera, but I dont want look lazy, but really I have lack of knowledge to imagine how make it work. 

Just say thanks by share knowledge, without  people like you hobbyist like me couldnt have tools to try to learn.

beomsu7, 2021-05-17:
I'm also beginner but I wanna give you some help and I'm px4 user

If the flight controller support the offboard mode, which means to control the drone with remote pc by sending target position, vel, acc etc, then you can do

The Fast-planner package has waypoint generator, so when you send waypoint(destination position), then it generate trajectory
Then you send the trajectory to mavros

For example
My drone has px4 and also px4 support offboard mode with mavros, so I can use offboard mode and control drone with remote pc. 
(1)  launch the mavros at the onboard pc of drone, run the camera(intel d435)
(2)  run the fast planner
(3)  set the waypoint(destination position)
(4) can get the trajectory from fast planner, and send the trajectory to mavros
[image]

Cavalletta98, 2022-11-14:
Can you pls provide the node that sends the traj to mavros?

sleepat-11, 2023-02-27:
hi, have you solved the problem? i want to know whether fast planner can be used on ardupilot?

sleepat-11, 2023-02-27:
@GarronLiu , Hi,I want to use RTK as UAV location,   for example I used longitude and latitude as waypoint position. I do not if it is possible for fast planner waypoint generator? can you explain the format of waypoint in fast planner? thanks!