HKUST-Aerial-Robotics/Fast-Planner #109: Is finer control possible with lower speed?

dulex123, 2021-05-20:
I'm having trouble tuning the planner with max_vel and max_acc. If I put the values too low the planned path becomes wierd/buggy (changing often too). Is there a way to exchange speed of flight for finer flight path?

Ideally I would like it to move very slow.

ZbyLGsc, 2021-07-28:
When applying a very low speed, you should increase 'search/max_tau'

3zuli, 2021-09-09:
Hello @ZbyLGsc, I have a similar problem with `kino_replan`. I want to use `max_vel = 0.5` and `max_acc = 1.0`, so following your previous advice I tried seting `max_tau = 1.0` and `resolution_astar = .01`.  The planner works fine in free space, but as soon as it encounters an obstacle the trajectory becomes very chaotic, sometimes it diverges and plans a trajectory to a completely different coordinate. If I further increase `max_tau` to a larger value like 2.0, it can't find a trajectory unless the waypoint is within about 1m from the current position. Can you give me some advice on how can I tune the parameters better?