HKUST-Aerial-Robotics/Fast-Planner #113: Can't make the map size larger

Kevinlibaba, 2021-07-27:
Hey all, I am applying the solution in my  own application and change the map size to 200*100*10 in kino_replan.launch. However, the planner said that: 
```
...
process[fast_planner_node-1]: started with pid [22489]
process[traj_server-2]: started with pid [22491]
process[waypoint_generator-3]: started with pid [22492]
process[odom_visualization-4]: started with pid [22503]
hit: 0.619039
miss: -0.619039
min log: -1.99243
max: 2.19722
thresh log: 1.38629
[ WARN] [1627410159.670548166]: [Traj server]: ready.
[fast_planner_node-1] process has died [pid 22489, exit code -9, cmd /home/rui/fast_ws/devel/lib/plan_manage/fast_planner_node /odom_world:=/airsim_node/drone_1/odom_local_ned /sdf_map/odom:=/airsim_node/drone_1/odom_local_ned /sdf_map/cloud:=/airsim_node/drone_1/points /sdf_map/pose:=/pcl_render_node/camera_pose /sdf_map/depth:=/pcl_render_node/depth __name:=fast_planner_node __log:=/home/rui/.ros/log/3029c27e-ef06-11eb-8be3-e86a647151f5/fast_planner_node-1.log].
log file: /home/rui/.ros/log/3029c27e-ef06-11eb-8be3-e86a647151f5/fast_planner_node-1*.log
```

The error messages didn't provide any helpful information and the log file can't find when i try to open the mentioned logs.
Have anyone or FAST team faced this problem?  I'd be grateful if you someone give me any recommendations.

ZbyLGsc, 2021-07-28:
Currently the map is not optimized for memory usage, so if you allocate a huge map, it may run out of all your memory. You may try to reduce the 'resolution' to relieve this problem.

Kevinlibaba, 2021-07-28:
@ZbyLGsc Wow!!! Thanks for your suggestion. It solved my question. I set the 'resolution' to 0.2 and crease the size of local_map in Rviz. Everything went well!

closed 2021-08-03 by Kevinlibaba

rsj007, 2022-03-11:
> @ZbyLGsc Wow!!! Thanks for your suggestion. It solved my question. I set the 'resolution' to 0.2 and increase the size of local_map in Rviz. Everything went well!

@Kevinlibaba Hello! Have you run the Fast Planner in airsim successfully? Can I get in touch with you? Thanks!!! My email is 736720216@qq.com

Kevinlibaba, 2022-03-11:
> > @ZbyLGsc Wow!!! Thanks for your suggestion. It solved my question. I set the 'resolution' to 0.2 and increase the size of local_map in Rviz. Everything went well!
> 
> @Kevinlibaba Hello! Have you run the Fast Planner in airsim successfully? Can I get in touch with you? Thanks!!! My email is [736720216@qq.com](mailto:736720216@qq.com)

Yes, I've done that. My email is lirui4217@icloud.com