HKUST-Aerial-Robotics/Fast-Planner #114: Gradient of end point cost may be wrong

BIRL-xu, 2021-07-28:
Hi, I'm learning the great project and found the gradient calculation about the end point cost:
https://github.com/HKUST-Aerial-Robotics/Fast-Planner/blob/2ddd7793eecd573dbb5b47e2c985aa06606df3cf/fast_planner/bspline_opt/src/bspline_optimizer.cpp#L291-L309
The cost function is f=dq^2, dq = 1/6*(q_3+4q_2+q_1), so the gradient should not be divided by 6.
E.g., the gradient for q_1, df/dq_1=df/dq * dq/dq_1=2 * dq * 1.

Is it right?

ZbyLGsc, 2021-07-28:
The derivative has no problem. 1/6 is obtained when applying the chain rule.

BIRL-xu, 2021-07-29:
@ZbyLGsc Yes, you are right! I'm sorry for the stupid question. Thank you for your reply.

BIRL-xu, 2021-07-29:
E.g., the gradient for q_1, df/dq_1=df/dq * dq/dq_1=2 * dq * 1/6.

closed 2021-07-29 by BIRL-xu