HKUST-Aerial-Robotics/Fast-Planner #115: Minimum Snap with Fast Planner

brunopinto900, 2021-08-05:
Hello, 

First of all, congratulations for such amazing work. Every one of HKUST papers on aerial robotics are very well written with lots of deep knowledge.

I am doing a project on autonomous drone racing, i.e. alpha pilot drone challenge. My approach so far (just to get started) is to feed the gate's waypoints to a minimum snap trajectory generator, then use the FastPlanner+ESDF map for local replanning & obstacle avoidance. Do you think this is a good approach? Or i might as well use straight lines rather than min.snap?

Thank you.

Kavin-Kailash, 2021-08-05:
Hello @brunopinto900. I have tried Fast Planner (simulations), and I believe that while it is a really effective and robust planner for agressive flight, it may not be the best choice for use as a local planner (which is what I think you are looking for), as it also performs map integration to enable multi-query type applications.

I had a look into this work https://github.com/ZJU-FAST-Lab/ego-planner , which is a cut down version of Fast-Planner which doesn't include the Map integration part, instead it only uses the real time sensor data for planning.

brunopinto900, 2021-08-05:
Hello @Kavin-Kailash. The EGO planner looks like more useful for my purpose than FastPlanner. Yes, i am looking for local planning. I will read the paper more in depth. Thank you. Are you working on something?

ZbyLGsc, 2021-08-05:
@brunopinto900 @Kavin-Kailash I have to clarify that the statement of  Kavin-Kailash is simply not true. EGO-Planner and Fast-Planner do the same task: generate local trajectories quickly. The difference is that EGO-planner does not build the ESDF, but only requires the occupancy map to perform gradient-based trajectory optimization.
It is even faster than Fast-Planner (it reduces computation time from ~5 ms to less than 1ms). However, the generated trajectories have higher jerk/snap.
So If you prefer faster computation, use EGO-Planner, and use Fast-Planner if you require smoother trajectory.

ZbyLGsc, 2021-08-05:
Also note that both of the planners perform raycasting to build the occupancy map. Actually the mapping module of EGO-Planner is inherited from Fast-Planner.

brunopinto900, 2021-08-05:
Well, i am implementing an autonomous drone racing on abandoned buildings, so i need quick replans, however i don't want non-optimal trajectories which my controller will have trouble follwing or simply take more time than it should. So what do you recommend?

ZbyLGsc, 2021-08-05:
I think all methods including ego-planner, kino_replan and topo_replan can be use to accomplish your task. Although the statistics of jerk/snap and computation time are different, practically there is no much difference. The replanning time ( several ms) and the slightly higher jerk/snap are really unconspicuous when you run the algorithms.
But in terms of implementation, the kino_replan may be slightly easier to use, since you only need to set the gate's waypoints repeatedly.

brunopinto900, 2021-08-05:
Thanks for your insights. I will read the paper associated with kino_replan and get to work.

brunopinto900, 2021-08-07:
@ZbyLGsc Do you think i could get equivalent results from using kinodynamic RRT* rather than kinodynamic A*? I know the later is the fastest in low dimensional space (which is the case). 

Other question: I am thinking of building a flight corridor and use MPC as the controller, instead of the SO(3) geometric controller. Is this a good approach?

I am asking all of this questions, because i am reading lots of papers, thinking of different approaches for my autonomous drone racing project.

Thank you for your time.