HKUST-Aerial-Robotics/Fast-Planner #116: Kinodynamic A*

jacketfffff, 2021-08-06:
For Kinodynamic A* algorithm, the expand-nodes method in the search method has become acceleration sampling and integral. So how to ensure the completeness of probability? Thank you.

brunopinto900, 2021-08-06:
Completness of probability in A*? A* is deterministic and optimal up to a resolution given an admissible heuristics (cost-to-go).

jacketfffff, 2021-08-07:
Thanks for your reply. I think the biggest advantage of A* is to find the optimal resolution if it has a resolution on the map. However, if the graph is no longer a fixed map and constructed by sampling, how to ensure that the existing solution can be found when the map would not be traversed entirely.

EhrazImam, 2021-08-19:
Have you used A* algorithm in path planning or not? If yes then pls tell where have you used it?
Please tell me as soon as possible. I will be very thankful to you