HKUST-Aerial-Robotics/Fast-Planner #117: Ubuntu 20 segmentation fault

brunopinto900, 2021-08-06:
Hello,

I launched 

When i give a goal with the 2D navigation goal plugin in RVIZ, its printed, on the terminal, the following:

riggered!
[TRIG]: from WAIT_TARGET to GEN_NEW_TRAJ
[kino replan]: -----------------------
start: -19   0   0, 0 0 0, 0 0 0
goal: -10.9901 -0.233514         1, 0 0 0
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Stack trace (most recent call last):
#8    Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#7    Object "/home/bruno/fast_plan_ws/devel/lib/libpath_searching.so", at 0x7f460ba8e570, in 
#6    Object "/home/bruno/fast_plan_ws/devel/lib/plan_manage/fast_planner_node", at 0x7f460bb208f9, in Eigen::internal::throw_std_bad_alloc()
#5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f460b13a798, in __cxa_throw
#4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f460b13a4e6, in std::terminate()
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f460b13a47b, in 
#2    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f460b12e950, in 
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f460ad45858, in abort
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f460ad660fb, in gsignal
Segmentation fault (Signal sent by the kernel [(nil)])
[fast_planner_node-1] process has died [pid 13388, exit code -11, cmd /home/bruno/fast_plan_ws/devel/lib/plan_manage/fast_planner_node /odom_world:=/state_ukf/odom /sdf_map/odom:=/state_ukf/odom /sdf_map/cloud:=/pcl_render_node/cloud /sdf_map/pose:=/pcl_render_node/camera_pose /sdf_map/depth:=/pcl_render_node/depth __name:=fast_planner_node __log:=/home/bruno/.ros/log/62741276-f6d0-11eb-9763-089798ca460b/fast_planner_node-1.log].
log file: /home/bruno/.ros/log/62741276-f6d0-11eb-9763-089798ca460b/fast_planner_node-1*.log

What could be wrong?

anyway-blows, 2021-11-01:
@iamrajee Thare are two more bugs in fast-planner codes, which are functions `interpolateTrilinear` and `evaluateEDTWithGrad`.

Their declarations both declaim a return value: [L69-L72](https://github.com/HKUST-Aerial-Robotics/Fast-Planner/blob/2ddd7793eecd573dbb5b47e2c985aa06606df3cf/fast_planner/plan_env/include/plan_env/edt_environment.h#L69-L72)
but their definition miss a return value:
 [L84-L105](https://github.com/HKUST-Aerial-Robotics/Fast-Planner/blob/2ddd7793eecd573dbb5b47e2c985aa06606df3cf/fast_planner/plan_env/src/edt_environment.cpp#L84-L105) 
[L107-L118](https://github.com/HKUST-Aerial-Robotics/Fast-Planner/blob/2ddd7793eecd573dbb5b47e2c985aa06606df3cf/fast_planner/plan_env/src/edt_environment.cpp#L107-L118) 

wish to be useful to you.

iamrajee, 2021-11-01:
@anyway-blows You're right, thank you for the insights. The weird part is that I have also tested the same code of Fast_Planner on ubuntu18.04 Melodic, where it works without any issue or error. It's only noetic where I'm getting this error(`open set empty, no path!`). Still not sure what might be causing this issue.

DavidePatria, 2022-01-17:
@iamrajee  have you been able to run it on noetic?

iamrajee, 2022-01-17:
Nopes! I eventually shifted to melodic *-*

eliabntt, 2022-02-22:
try #85 try this [comment](https://github.com/HKUST-Aerial-Robotics/Fast-Planner/issues/85#issuecomment-1047543010)

ba1Sta, 2023-05-21:
> @iamrajee Thare are two more bugs in fast-planner codes, which are functions `interpolateTrilinear` and `evaluateEDTWithGrad`.
> 
> Their declarations both declaim a return value: [L69-L72](https://github.com/HKUST-Aerial-Robotics/Fast-Planner/blob/2ddd7793eecd573dbb5b47e2c985aa06606df3cf/fast_planner/plan_env/include/plan_env/edt_environment.h#L69-L72) but their definition miss a return value: [L84-L105](https://github.com/HKUST-Aerial-Robotics/Fast-Planner/blob/2ddd7793eecd573dbb5b47e2c985aa06606df3cf/fast_planner/plan_env/src/edt_environment.cpp#L84-L105) [L107-L118](https://github.com/HKUST-Aerial-Robotics/Fast-Planner/blob/2ddd7793eecd573dbb5b47e2c985aa06606df3cf/fast_planner/plan_env/src/edt_environment.cpp#L107-L118)
> 
> their returns are unnecessary, so I changed them to return `void`.
> 
> wish to be useful to you.

Hi @iamrajee  I encountered the same problem: open set empty, no path.... After follow the instruction from @anyway-blows . I successfully figure that out. Here is my modification:
*In the edt_environment.h, replace "pair<double, Eigen::Vector3d>" with void in both line 69 and line 71. After modification:
```
/**
* This file is part of Fast-Planner.
*
* Copyright 2019 Boyu Zhou, Aerial Robotics Group, Hong Kong University of Science and Technology, <uav.ust.hk>
* Developed by Boyu Zhou <bzhouai at connect dot ust dot hk>, <uv dot boyuzhou at gmail dot com>
* for more information see <https://github.com/HKUST-Aerial-Robotics/Fast-Planner>.
* If you use this code, please cite the respective publications as
* listed on the above website.
*
* Fast-Planner is free software: you can redistribute it and/or modify
* it under the terms of the GNU Lesser General Public License as published by
* the Free Software Foundation, either version 3 of the License, or
* (at your option) any later version.
*
* Fast-Planner is distributed in the hope that it will be useful,
* but WITHOUT ANY WARRANTY; without even the implied warranty of
* MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
* GNU General Public License for more details.
*
* You should have received a copy of the GNU Lesser General Public License
* along with Fast-Planner. If not, see <http://www.gnu.org/licenses/>.
*/

#ifndef _EDT_ENVIRONMENT_H_
#define _EDT_ENVIRONMENT_H_

#include <Eigen/Eigen>
#include <geometry_msgs/PoseStamped.h>
#include <iostream>
#include <ros/ros.h>
#include <utility>

#include <plan_env/obj_predictor.h>
#include <plan_env/sdf_map.h>

using std::cout;
using std::endl;
using std::list;
using std::pair;
using std::shared_ptr;
using std::unique_ptr;
using std::vector;

namespace fast_planner {
class EDTEnvironment {
private:
  /* data */
  ObjPrediction obj_prediction_;
  ObjScale obj_scale_;
  double resolution_inv_;
  double distToBox(int idx, const Eigen::Vector3d& pos, const double& time);
  double minDistToAllBox(const Eigen::Vector3d& pos, const double& time);

public:
  EDTEnvironment(/* args */) {
  }
  ~EDTEnvironment() {
  }

[... 134 lines cut ...]
  sdf_map_->getSurroundPts(pos, sur_pts, diff);

  double dists[2][2][2];
  getSurroundDistance(sur_pts, dists);

  interpolateTrilinear(dists, diff, dist, grad);
}

double EDTEnvironment::evaluateCoarseEDT(Eigen::Vector3d& pos, double time) {
  double d1 = sdf_map_->getDistance(pos);
  if (time < 0.0) {
    return d1;
  } else {
    double d2 = minDistToAllBox(pos, time);
    return min(d1, d2);
  }
}

// EDTEnvironment::
}  // namespace fast_planner
```
* cd to the workspace folder and catkin_make
```
catkin_make
```

ba1Sta, 2023-05-21:
> @iamrajee Thare are two more bugs in fast-planner codes, which are functions `interpolateTrilinear` and `evaluateEDTWithGrad`.
>
> Their declarations both declaim a return value: [...]

Thanks! This really helped me out! Previous I encounter the same problem which is mentioned by @iamrajee . Now the simulation is good to go. Also, thank you teams from HKUST-Aerial-Robotics! I am also willing to keep contributing to this repo. If anyone still in the trouble working with Ubuntu 20.04 and ROS-Noetic. Please feel free to contact me anytime!

yanzw1, 2023-09-14:
> Thanks! This really helped me out! Previous I encounter the same problem which is mentioned by @iamrajee . Now the simulation is good to go. Also, thank you teams from HKUST-Aerial-Robotics! I am [...]

I met the same problem which is mentioned by @iamrajee .how could i fix it thanks！！@ba1Sta