HKUST-Aerial-Robotics/Fast-Planner #121: How to save more ESDF map

Kevinlibaba, 2021-08-29:
Hey Zhou, How can I save more map as shown in Figure 1 ? I have tried your comments in #60, #70 and #35, but not effect on map in RVIZ. Do you have any idea to solve that ?
P.S. I have already set `sdf_map/local_map_margin` to 500 and set `publishMapInflate(true)`, as well as commented `clearAndInflateLocalMap`.

DeveloperFuloo, 2023-03-17:
Besides comment line 652-722 in sdf_map.cpp, i comment line 729-733 line in sdf_map.cpp that commented `clear outdated data`,it will save the global map

eldenring998, 2023-03-24:
> Hey Zhou, How can I save more map as shown in Figure 1 ? I have tried your comments in #60, #70 and #35, but not effect on map in RVIZ. Do you have any idea to solve that ? P.S. I have already set [...]

Hello! I encountered the same problem as you, I modified all the places you mentioned, but the content displayed in rviz did not change. The clearAndInflateLocalMap function will not be called during simulation, so it is useless to modify it. Modifying sdf_map/local_map_margin does not work either, because it is the field of view size that limits the size of the ESDF map, not this parameter. The update of the map is done in the updateESDF3d function. This function receives the point cloud information and updates it at high speed. It only publishes the map composed of the current point cloud. As long as the point cloud changes, the map will change accordingly, and the previous frame cannot be retained. map information.
All in all, I tried many methods and still failed to build a global map. Did you succeed? Looking forward to your reply.

eldenring998, 2023-03-24:
> Hey Zhou, How can I save more map as shown in Figure 1 ? I have tried your comments in #60, #70 and #35, but not effect on map in RVIZ. Do you have any idea to solve that ? P.S. I have already set [...]

hello，I met the same problem as you, did you solve it?

Kevinlibaba, 2023-03-24:
> hello，I met the same problem as you, did you solve it?

Hi @eldenring998, it didn't run for me, sorry I can't help you.

eldenring998, 2023-03-26:
> Hi @eldenring998, it didn't run for me, sorry I can't help you.

hello，I seem to have succeeded. In addition to the operations you mentioned above, you also need to delete lines 876-877 in sdf_map.cpp. In the program, occupancy_buffer_inflate_ stores point cloud data, and the code on lines 876-877 will put occupancy_buffer_inflate_ in the array , the point cloud outside the update range is deleted.

DeveloperFuloo, 2023-03-28:
> > > > Hey Zhou, How can I save more map as shown in Figure 1 ? I have tried your comments in #60, #70 and #35, but not effect on map in RVIZ. Do you have any idea to solve that ? P.S. I have already set `sdf_map/local_map_margin` to 500 and set `publishMapInflate(true)`, as well as commented `clearAndInflateLocalMap`.
> > > 
> > > 
> > > hello，I met the same problem as you, did you solve it?
> > 
> > 
> > Hi @eldenring998, it didn't run for me, sorry I can't help you.
> 
> hello，I seem to have succeeded. In addition to the operations you mentioned above, you also need to delete lines 876-877 in sdf_map.cpp. In the program, occupancy_buffer_inflate_ stores point cloud data, and the code on lines 876-877 will delete the point cloud in occupancy_buffer_inflate that is outside the update range.

hi， @eldenring998 ,let me confirm the location of the code you mentioned,  do you mean the `resetBuffer` function?

eldenring998, 2023-03-28:
> hi， @eldenring998 ,let me confirm the location of the code you mentioned, do you mean the `resetBuffer` function?

yes，As long as the resetBuffer function is prevented from deleting data, all map data will be retained