HKUST-Aerial-Robotics/Fast-Planner #128: Migration from catkin_make to catkin build

Cristian-wp, 2021-09-10:
``Hello, I have already test your planner and is really awesome.
But for my project I need to compile my workspace with catkin build instead of catkin_make.
Have someone already try this or maybe there is a version that work with catkin build?

When I try to compile it I get this error:
```
Could NOT find quadrotor_msgs (missing: quadrotor_msgs_DIR)
-- Could not find the required component 'quadrotor_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "quadrotor_msgs"
  with any of the following names:

    quadrotor_msgsConfig.cmake
    quadrotor_msgs-config.cmake

  Add the installation prefix of "quadrotor_msgs" to CMAKE_PREFIX_PATH or set
  "quadrotor_msgs_DIR" to a directory containing one of the above files.  If
  "quadrotor_msgs" provides a separate development package or SDK, be sure it
  has been installed.
```

I have already try to compile and install the packages with this repository, but it did not work:
[https://github.com/HKUST-Aerial-Robotics/plan_utils](url)

If I modify the CMAKE_PREFIX_PATH as suggested, it find the quadrotor_msgs, but it can not fine the basic packages like roscpp anymore...

Can someone help me please?

3zuli, 2021-09-10:
Hello, porting to catkin build has been discussed previously, for example in #18 and #81. Following those two issues I managed to port it to catkin build and I made a [summary patch](https://gist.github.com/3zuli/f76936b75ef6aef3a203a1402f559d5c) with the changes that I had to do in order to make it work. I also had similar compile issues with missing Fast-Planner packages/libraries, I had to build those packages one by one (`catkin build  quadrotor_msgs`) until the entire `catkin build` succeeded. I hope this helps.

Cristian-wp, 2021-09-10:
Thank you very much, I try now and let you now!

Cristian-wp, 2021-09-10:
It works! Thanks a lot!