HKUST-Aerial-Robotics/Fast-Planner #129: 使用depth_topic还是cloud_topic？

sbdjyl, 2021-09-24:
在kino_replan.launch中有这样两个注释：
  
    
    

以及
    
    
    

是不是可以认为在使用时depth_topic和cloud_topic只需要其中之一？
但是当我注释掉cloud_topic后，运行会报错：

included file [/home/uav2/catkin_lh/src/Fast-Planner/fast_planner/plan_manage/launch/kino_algorithm.xml] requires the 'cloud_topic' arg to be set

我该怎么样设置才能把它应用在实际飞行中呢？

zyhphd, 2021-12-08:
你报错的原因是xml文件中有cloud_topic，所以你需要在launch中需要定义。你可以尝试不注释，空值就行了。

mtncgj123, 2022-09-15:
请问你的问题解决了吗？ 我现在跑fast planner 实际地图一直出不来

LiuL0ngl0ng, 2022-12-06:
> 请问你的问题解决了吗？ 我现在跑fast planner 实际地图一直出不来

你的问题解决了吗？我也无法出实时地图？是Opencv的问题吗？

Zyhlibrary, 2022-12-06:
> 首先你们需要确认topic输入是否正确；其次确认topic里是否有数据；另外，要同时设置camera_pos的topic

> > 请问你的问题解决了吗？ 我现在跑fast planner 实际地图一直出不来
> 
> 你的问题解决了吗？我也无法出实时地图？是Opencv的问题吗？

首先你们需要确认topic输入是否正确；其次确认topic里是否有数据；另外，要同时设置camera_pos的topic

sirius-zyj, 2024-11-21:
首先确保cloud_topic和depth topic只有一个，即你想使用哪个就订阅哪个
其次使用depth topic话题时还得同时订阅camera_pos，这两的时间戳也得同步，不然无法生成点云数据