HKUST-Aerial-Robotics/Fast-Planner #136: Point cloud orientation problem

bhaskar-glitch, 2021-12-17:
Hi,
Im using realsense d435i and facing an issue with the point cloud of realsense of topic /camera/depth/color/points
in orientation of pointcloud is -y is upward...the problem is -y is facing toward z axis as in this picture
[image]

Please help

luccagandra, 2021-12-29:
You can create a static transform from base_link to camera_link in your launch file. Ex: 
<node pkg="tf2_ros" type="static_transform_publisher" name="link1_broadcaster" args="0 0 0 -1.57 0 -1.57 base_link camera_link"/>

luccagandra, 2021-12-29:
You can create a transform from base_link to camera link using static transform. Mine looks like that: 
node pkg="tf2_ros" type="static_transform_publisher" name="link1_broadcaster" args="0 0 0 -1.57 0 -1.57 base_link camera_link

Xchegg, 2023-05-08:
Have you solved this issue?

DavidePatria, 2023-05-08:
I think I had i solved [here](https://github.com/DavidePatria/rtabmap_fast_planner_px4/blob/ef3dd1ee03ab4b041341d9dfc4fc56e42cbe32a3/rtabmap_drone/launch/spawn_iris_depth.launch#LL69C1-L69C144), try and report.
the environment was a simple iris drone as described in that repo

Xchegg, 2023-05-10:
I found that the point cloud in RVIZ is inverted, and doing a symmetrical transformation on the Z axis and then moving the Z axis can solve this problem.

Xchegg, 2023-05-17:
I have already solved this problem. Do not use the camera's point cloud as input. I use the octree map package to build a map, and then use the point cloud of the octree map as input to solve this problem.