HKUST-Aerial-Robotics/Fast-Planner #137: Huge delay in Point Cloud

bhaskar-glitch, 2021-12-19:
Hi,
I'm using the depth image from Realsense d435i camera for the point cloud(/camera/depth/img_rect_raw) in Fast planner but the point cloud is having a huge delay of 5-6 seconds,it's a big delay for drones and it may crash.Is this a problem of fast planner(converting depth to pcl) or rviz(publish rate)? PLease help regarding the issue
**The Point cloud is like this**
[image]

**Rqt graph is**
[image]

ziyu1018, 2022-01-11:
Is this map an ESDF map?

minhdov, 2024-05-21:
Are there any updates on this?

bhaskar-glitch, 2024-06-20:
> Are there any updates on this?

It was because of the topic hz, which was slow in my case

devkishan920, 2026-01-07:
> > Are there any updates on this?
> 
> It was because of the topic hz, which was slow in my case

Have you applied above to real drone.
I am also trying to implement same using d435i. Right now is am passing /camera/depth/image_rect_raw to /fast/planner_node. But, I can't pass the environment (streamed from d435i) to rviz simulation. 

Any help would be greatly appreciated!!!!