HKUST-Aerial-Robotics/Fast-Planner #138: Issues with using catkin build

Kaijun101, 2022-01-13:
Hi, I tried to migrate to catkin build according to the pull request #74, however I am faced with multiple warnings that is unresolved. I have attached the warnings for reference too.

Warnings   << map_generator:cmake /home/jetson/planner_ws/logs/map_generator/build.cmake.000.log                                                                                                           
** WARNING ** io features related to ensenso will be disabled
** WARNING ** io features related to davidSDK will be disabled
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
** WARNING ** visualization features related to ensenso will be disabled
** WARNING ** visualization features related to davidSDK will be disabled
** WARNING ** visualization features related to dssdk will be disabled
** WARNING ** visualization features related to rssdk will be disabled
cd /home/jetson/planner_ws/build/map_generator; catkin build --get-env map_generator | catkin env -si  /usr/bin/cmake /home/jetson/planner_ws/src/Fast-Planner/uav_simulator/map_generator --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/jetson/planner_ws/devel/.private/map_generator -DCMAKE_INSTALL_PREFIX=/home/jetson/planner_ws/install; cd -

Warnings   << so3_disturbance_generator:cmake /home/jetson/planner_ws/logs/so3_disturbance_generator/build.cmake.000.log                                                                                   
CMake Warning (dev) at CMakeLists.txt:70 (add_dependencies):
  Policy CMP0046 is not set: Error on non-existent dependency in
  add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  The dependency target "so3_disturbance_generator_gencfg" of target
  "so3_disturbance_generator" does not exist.
This warning is for project developers.  Use -Wno-dev to suppress it.

Warnings   << local_sensing_node:cmake /home/jetson/planner_ws/logs/local_sensing_node/build.cmake.000.log                                                                                                 
CMake Warning at /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'Eigen' but neither 'Eigen_INCLUDE_DIRS' nor
  'Eigen_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:35 (catkin_package)

Warnings   << so3_quadrotor_simulator:cmake /home/jetson/planner_ws/logs/so3_quadrotor_simulator/build.cmake.000.log                                                                                       
CMake Warning at /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'Eigen3' but neither 'Eigen3_INCLUDE_DIRS' nor
  'Eigen3_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:25 (catkin_package)

CMake Warning at /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'system_lib' but neither
  'system_lib_INCLUDE_DIRS' nor 'system_lib_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:25 (catkin_package)

Warnings   << rviz_plugins:cmake /home/jetson/planner_ws/logs/rviz_plugins/build.cmake.000.log                                                                                                             
CMake Warning at /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'system_lib' but neither
  'system_lib_INCLUDE_DIRS' nor 'system_lib_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:12 (catkin_package)

Warnings   << traj_utils:cmake /home/jetson/planner_ws/logs/traj_utils/build.cmake.000.log                                                                                                                 
** WARNING ** io features related to ensenso will be disabled
** WARNING ** io features related to davidSDK will be disabled
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
** WARNING ** visualization features related to ensenso will be disabled
** WARNING ** visualization features related to davidSDK will be disabled
** WARNING ** visualization features related to dssdk will be disabled
** WARNING ** visualization features related to rssdk will be disabled
cd /home/jetson/planner_ws/build/traj_utils; catkin build --get-env traj_utils | catkin env -si  /usr/bin/cmake /home/jetson/planner_ws/src/Fast-Planner/fast_planner/traj_utils --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/jetson/planner_ws/devel/.private/traj_utils -DCMAKE_INSTALL_PREFIX=/home/jetson/planner_ws/install; cd -

Warnings   << plan_manage:cmake /home/jetson/planner_ws/logs/plan_manage/build.cmake.000.log                                                                                                               
** WARNING ** io features related to ensenso will be disabled
** WARNING ** io features related to davidSDK will be disabled
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
** WARNING ** visualization features related to ensenso will be disabled
** WARNING ** visualization features related to davidSDK will be disabled
** WARNING ** visualization features related to dssdk will be disabled
** WARNING ** visualization features related to rssdk will be disabled
cd /home/jetson/planner_ws/build/plan_manage; catkin build --get-env plan_manage | catkin env -si  /usr/bin/cmake /home/jetson/planner_ws/src/Fast-Planner/fast_planner/plan_manage --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/jetson/planner_ws/devel/.private/plan_manage -DCMAKE_INSTALL_PREFIX=/home/jetson/planner_ws/install; cd -

DavidePatria, 2022-01-17:
the problem here is that the packages are not built in the correct order that would satisfy the requirements. what I tried is building singles packages with `catkin build` until the compilation succeeded and I got another error that doesn't seem to be fixable by me, related to problems with the link of a library present in one of the packages. I'm staying with `catkin_make` for now.

Kaijun101, 2022-01-17:
Hi @DavidePatria thanks for your reply. By any chance would you know the correct order for me to use catkin build ?

DavidePatria, 2022-01-17:
no, I'm using `catkin_make`

closed 2022-02-04 by Kaijun101