HKUST-Aerial-Robotics/Fast-Planner #140: intrinsic params of the depth camera

JonathanLJH, 2022-02-13:
I want to know where the data came from
I use a t265 camera
But I didn't see relevant data during calibration
[image]

closed 2022-02-14 by JonathanLJH

greymaner, 2022-03-17:
I also want to know how to calibrate depth camera (realsense d435i) intrinsic, do you have solved this problem?

FPSychotic, 2022-03-17:
You both will find such data in the camera_info topics of realsense_camera2  node.
You need print it with rostopic  echo and the data should be in the K section I think

greymaner, 2022-03-21:
@FPSychotic  thanks for your reply,but I wonder if the /d435/depth/camera_info is the correct parameters, or is this topic read from config files rather than calibrated online . And when I use the printed params in the launch file, the positions of obstacles in voxel map seem not accurate .

FPSychotic, 2022-04-13:
> @FPSychotic thanks for your reply,but I wonder if the /d435/depth/camera_info is the correct parameters, or is this topic read from config files rather than calibrated online . And when I use the printed params in the launch file, the positions of obstacles in voxel map seem not accurate .

that calibration data are correct as it is calibrated from factory and recorded in the eeprom. Only in case you needed and did change such calibration with the realsense tools you could have bad calibration data.